marq24/ha-senec-v3 #109: Fetching Data from Senec V2.1 broken since 2024.10.0

neubi4, 2024-11-02:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hi,

since Updating from 2024.9.0 to 2024.11.0, the integration is broken for the local API and only reports "Einrichtung fehlgeschlagen, wird erneut versucht: None" for my Senec V2.1.

I have enabled debug logging, see attached log.

I tried 2024.10.0 but it is also broken. 2024.9.0 works, so something must be changed between .9 and .10.

My evcc still successfully pulls data from my senec, to the senec should not be the issue.

### Add your DEBUG log output

```shell
2024-11-02 13:14:29.124 INFO (MainThread) [custom_components.senec] Starting SENEC.Home Integration 'senec' with interval:0:01:00 - ConfigEntry: {'created_at': '1970-01-01T00:00:00+00:00', 'data': {'name': 'senec', 'host': '<MASKED>', 'use_https': True, 'scan_interval': 60, 'type': 'senec', 'dtype_int': 15, 'dtype': 'SENEC Main-Unit', 'dname': 'Senec Home V2.1 1ph / Lithium | SENEC.Inverter V2.1', 'dserial': '<MASKED>', 'version': 'App:0825 FW:2307 NPU-Image:2309(v10)'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'senec', 'entry_id': '0c243a7bab78c7ae0c08789f46097085', 'minor_version': 1, 'modified_at': '2024-11-02T11:51:47.970561+00:00', 'options': {'name': 'senec', 'host': '<MASKED>', 'scan_interval': 60, 'ignore_system_state': False}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'senec', 'unique_id': None, 'version': 1}
2024-11-02 13:14:29.124 INFO (MainThread) [custom_components.senec] ***** QUERY_BMS-DATA ********
2024-11-02 13:14:29.125 INFO (MainThread) [custom_components.senec] ***** QUERY_FANSPEED-DATA ********
2024-11-02 13:14:29.125 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting Senec lala.cgi integration... for host: '192.168.251.135' with options: {'ignore_system_state': False, 'query_wallbox_data': False, 'query_wallbox_data_via_app_api': False, 'query_bms_data': True, 'query_fan_data': True, 'query_sockets_data': False}
2024-11-02 13:14:29.395 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-11-02 13:14:29.395 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.270 seconds (success: False)
```

marq24, 2024-11-02:
Martin is it possible for you to send me the content (via eMail) of the lala.cpi POST request - so that I can debug this locally?!

neubi4, 2024-11-02:
I hope i used the correct post data, tried to figure it out from here: https://github.com/marq24/ha-senec-v3/blob/2438c671363476bb4b9ce008f6359cb530b14bb2/custom_components/senec/pysenec_ha/__init__.py#L1520

`curl -XPOST -d '{"ENERGY": {"STAT_STATE": "", "GUI_GRID_POW": "", "GUI_HOUSE_POW": "", "GUI_INVERTER_POWER": "", "GUI_BAT_DATA_FUEL_CHARGE": "", "GUI_BAT_DATA_POWER": "", "GUI_BAT_DATA_VOLTAGE": "", "GUI_BAT_DATA_CURRENT": "", "SAFE_CHARGE_RUNNING": "", "LI_STORAGE_MODE_RUNNING": ""}, "TEMPMEASURE": {"BATTERY_TEMP": "", "CASE_TEMP": "", "MCU_TEMP": ""}, "PV1": {"POWER_RATIO": "", "POWER_RATIO_L1": "", "POWER_RATIO_L2": "", "POWER_RATIO_L3": "", "MPP_VOL": "", "MPP_CUR": "", "MPP_POWER": ""}, "PWR_UNIT": {"POWER_L1": "", "POWER_L2": "", "POWER_L3": ""}, "PM1OBJ1": {"FREQ": "", "U_AC": "", "I_AC": "", "P_AC": "", "P_TOTAL": ""}, "PM1OBJ2": {"FREQ": "", "U_AC": "", "I_AC": "", "P_AC": "", "P_TOTAL": ""}, "FAN_SPEED": {}, "BMS": {"CELL_TEMPERATURES_MODULE_A": "", "CELL_TEMPERATURES_MODULE_B": "", "CELL_TEMPERATURES_MODULE_C": "", "CELL_TEMPERATURES_MODULE_D": "", "CELL_VOLTAGES_MODULE_A": "", "CELL_VOLTAGES_MODULE_B": "", "CELL_VOLTAGES_MODULE_C": "", "CELL_VOLTAGES_MODULE_D": "", "CURRENT": "", "VOLTAGE": "", "SOC": "", "SOH": "", "CYCLES": "", "MODULES_CONFIGURED": ""}}' -k https://192.168.251.135/lala.cgi`

```
{
  "ENERGY": {
    "STAT_STATE": "u8_0E",
    "GUI_GRID_POW": "fl_C1C68F5C",
    "GUI_HOUSE_POW": "fl_440D857A",
    "GUI_INVERTER_POWER": "fl_44814000",
    "GUI_BAT_DATA_FUEL_CHARGE": "fl_4262433B",
    "GUI_BAT_DATA_POWER": "fl_43DD8C16",
    "GUI_BAT_DATA_VOLTAGE": "fl_424F8937",
    "GUI_BAT_DATA_CURRENT": "fl_4108A3D7",
    "SAFE_CHARGE_RUNNING": "u8_00",
    "LI_STORAGE_MODE_RUNNING": "u8_00"
  },
  "TEMPMEASURE": {
    "BATTERY_TEMP": "fl_420C0000",
    "CASE_TEMP": "fl_421F3AAB",
    "MCU_TEMP": "fl_4262524A"
  },
  "PV1": {
    "POWER_RATIO": "fl_42C80000",
    "POWER_RATIO_L1": "fl_42C80000",
    "POWER_RATIO_L2": "fl_42C80000",
    "POWER_RATIO_L3": "fl_42C80000",
    "MPP_VOL": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "MPP_CUR": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "MPP_POWER": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ]
  },
  "PWR_UNIT": {
    "POWER_L1": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "POWER_L2": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "POWER_L3": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ]
  },
  "PM1OBJ1": {
    "FREQ": "fl_4247D70A",
    "U_AC": [
      "fl_436EE667",
      "fl_4371199A",
      "fl_436ECCCD"
    ],
    "I_AC": [
      "fl_3FDEB852",
      "fl_3FB0A3D7",
      "fl_3FB47AE1"
    ],
    "P_AC": [
      "fl_4388BD70",
      "fl_C27A6666",
      "fl_C36BB0A3"
    ],
    "P_TOTAL": "fl_C1C68F5C"
  },
  "PM1OBJ2": {
    "FREQ": "fl_00000000",
    "U_AC": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "I_AC": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "P_AC": [
      "fl_00000000",
      "fl_00000000",
      "fl_00000000"
    ],
    "P_TOTAL": "fl_00000000"
  },
  "FAN_SPEED": {
    "OBJECT_NOT_FOUND": ""
  },
  "BMS": {
    "CELL_TEMPERATURES_MODULE_A": [
      "i8_20",
      "i8_21",
      "i8_22",
      "i8_22",
      "i8_22",
      "i8_22"
    ],
    "CELL_TEMPERATURES_MODULE_B": [
      "i8_21",
      "i8_21",
      "i8_22",
      "i8_22",
      "i8_23",
      "i8_22"
    ],
    "CELL_TEMPERATURES_MODULE_C": [
      "i8_22",
      "i8_22",
      "i8_23",
      "i8_22",
      "i8_23",
      "i8_22"
    ],
    "CELL_TEMPERATURES_MODULE_D": [
      "i8_00",
      "i8_00",
      "i8_00",
      "i8_00",
      "i8_00",
      "i8_00"
    ],
    "CELL_VOLTAGES_MODULE_A": [
      "u1_0E7B",
      "u1_0E7C",
      "u1_0E7B",
      "u1_0E79",
      "u1_0E79",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E7B",
      "u1_0E7B",
      "u1_0E7A",
      "u1_0E7B",
      "u1_0E7B",
      "u1_0E7B",
      "u1_0E7A"
    ],
    "CELL_VOLTAGES_MODULE_B": [
      "u1_0E7B",
      "u1_0E78",
      "u1_0E78",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E7B",
      "u1_0E7A",
      "u1_0E7C",
      "u1_0E7B",
      "u1_0E7C",
      "u1_0E79",
      "u1_0E79",
      "u1_0E7B"
    ],
    "CELL_VOLTAGES_MODULE_C": [
      "u1_0E7B",
      "u1_0E7B",
      "u1_0E78",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E79",
      "u1_0E78",
      "u1_0E79",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E7A",
      "u1_0E77",
      "u1_0E7A"
    ],
    "CELL_VOLTAGES_MODULE_D": [
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000",
      "u1_0000"
    ],
    "CURRENT": [
      "fl_4030A3D7",
      "fl_403A3D71",
      "fl_4037AE14",
      "fl_00000000"
    ],
    "VOLTAGE": [
      "fl_424F9062",
      "fl_424F8B44",
      "fl_424F820C",
      "fl_00000000"
    ],
    "SOC": [
      "u1_0039",
      "u1_0039",
      "u1_0039",
      "u1_0000"
    ],
    "SOH": [
      "u8_63",
      "u8_63",
      "u8_63",
      "u8_00"
    ],
    "CYCLES": [
      "u1_0064",
      "u1_0064",
      "u1_0064",
      "u1_0000"
    ],
    "MODULES_CONFIGURED": "u8_03"
  }
}
```

marq24, 2024-11-02:
fix is on the way...

marq24, 2024-11-02:
https://github.com/marq24/ha-senec-v3/releases/tag/2024.11.1

neubi4, 2024-11-02:
Works, thank you very much :)

marq24, 2024-11-02:
just in case (solltest Du es noch nicht gesehen haben) - wenn Du Deinen Senec nicht zusätzlich mit den evcc requests belasten möchtest... -> https://github.com/marq24/ha-senec-v3/blob/master/docs/EVCC.md