marq24/ha-senec-v3 #110: Inverter aren’t working anymore.

Cyrelion, 2024-11-03:
### With which version of the integration you have an issue?

Internal inverter build into SENEC.Home V3 hybrid/hybrid duo

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Inverters weren’t working anymore so I’ve removed the devices and tried to readd them but it doesn’t let me to add the devices and throws an unknown error.

### Add your DEBUG log output

```shell
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 477, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 559, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 370, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 417, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 520, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/config_flow.py", line 270, in async_step_system
    if await self._test_connection_inverter(host_entry):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/config_flow.py", line 152, in _test_connection_inverter
    await inverter_client.update_version()
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 2105, in update_version
    await self.read_inverter_version()
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 2119, in read_inverter_version
    for a_entry in a_dict.keys():
                   ^^^^^^^^^^^
AttributeError: 'list' object has no attribute 'keys'
```

KneterPeter, 2024-11-04:
Die Inverter haben jetzt ein Webinterface mit vielen Informationen! 
[image]

marq24, 2024-11-04:
> Die Inverter haben jetzt ein Webinterface mit vielen Informationen! - Die Inverter haben 'schon immer' ein eigenes Webinterface... Was aber eben nur dann zugänglich ist, wenn man auch das LAN Kabel reinsteck...

marq24, 2024-11-04:
@Cyrelion - bist Du Dir sicher, das Du die AKTUELLSTE Version der Integration installiert hast?!
-> https://github.com/marq24/ha-senec-v3/releases/tag/2024.11.1

Cyrelion, 2024-11-04:
Ja, neuste Version ist installiert.
[image]

Cyrelion, 2024-11-04:
Hab auch gerade noch mal nen redownload gemacht und neu gestartet. Ich kann die weiterhin nicht erneut hinzufügen.

marq24, 2024-11-04:
auf welcher HA Version bist Du denn?

Cyrelion, 2024-11-04:
Ebenfalls die neuste Release-Version 2024.10.4

marq24, 2024-11-04:
Danke Für Deine Geduld

marq24, 2024-11-04:
Kannst Du bitte testen, ob dieses PreRelease jetzt funktioniert - mein zweiter Inverter ist Aktuell leider defekt - deswegen kann ich das selbst gar nicht testen :-/

https://github.com/marq24/ha-senec-v3/releases/tag/2024.11.2

closed 2024-11-04 by marq24