marq24/ha-senec-v3 #112: Mehrere Anlagen in einem Senec-Konto

be1001, 2024-11-19:
### With which version of the integration you have an issue?

SENEC.Home V4/SENEC.Home V4 hybrid

### Multiple Systems Environment?

- [X] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hallo, 
ich habe unter meinem Senec-Konto aktuell 3 Speicher. 
1x V2.1 AC Speicher
1x P4 Hybrid Speicher
1x P4 AC Speicher

wenn ich die Integration starte findet das System aktuell nur den V2.1 Speicher, der demnächst rausfliegt

### Add your DEBUG log output

```shell
2024-11-19 13:50:16.962 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration pfsense which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.963 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration waste_collection_schedule which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.964 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.965 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration senec which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.965 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.966 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration webrtc which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.966 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration garo_wallbox which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.967 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration mikrotik_router which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.968 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration alfen_wallbox which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:16.968 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration volkswagen_we_connect_id which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-19 13:50:20.668 WARNING (ImportExecutor_0) [homeassistant.helpers.typing] HomeAssistantType was used from webrtc, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'webrtc' custom integration
2024-11-19 13:50:20.671 WARNING (ImportExecutor_0) [homeassistant.helpers.typing] ServiceCallType was used from webrtc, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.ServiceCall instead, please report it to the author of the 'webrtc' custom integration
2024-11-19 13:50:20.686 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'webrtc' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/webrtc/video-rtc.js", "/config/custom_components/webrtc/www/video-rtc.js", True)])`; This function will be removed in 2025.7 at custom_components/webrtc/__init__.py, line 67: hass.http.register_static_path("/webrtc/" + name, str(path / name)), please create a bug report at https://github.com/AlexxIT/WebRTC/issues
2024-11-19 13:50:20.694 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'webrtc' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/webrtc/embed", "/config/custom_components/webrtc/www/embed.html", True)])`; This function will be removed in 2025.7 at custom_components/webrtc/__init__.py, line 74: hass.http.register_static_path("/webrtc/embed", str(path / "embed.html")), please create a bug report at https://github.com/AlexxIT/WebRTC/issues
2024-11-19 13:50:26.916 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 377, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 563, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Expected CRLF after version:

    b'POST /api/webhook/onvif000fc915bf69 HTTP/1.1 '
                                                  ^
2024-11-19 13:50:26.951 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Einspeisung VNB 164 ein/aus' failed to setup triggers and has been disabled: Unknown device '4dc89c0f38650b7cff1dc9bba0283780'
2024-11-19 13:50:26.986 WARNING (SyncWorker_12) [weconnect] /vehicles/WVGZZZE27RE004246/domains/automation/chargingProfiles/profiles/1/options: Unknown attribute usePrivateCurrentEnabled with value True
2024-11-19 13:50:27.008 WARNING (SyncWorker_12) [weconnect] /vehicles/WVGZZZE27RE004246/domains/charging/chargingStatus: Unknown attribute chargingScenario with value off
2024-11-19 13:50:27.011 WARNING (SyncWorker_12) [weconnect] /vehicles/WVGZZZE27RE004246/domains/chargingProfiles/chargingProfilesStatus/profiles/1/options: Unknown attribute usePrivateCurrentEnabled with value True
2024-11-19 13:50:27.019 WARNING (SyncWorker_12) [weconnect] /vehicles/WVGZZZE27RE004246: Unknown attribute climatisationTemperatureOutside with value {'error': {'message': 'Bad Gateway', 'errorTimeStamp': '2024-11-19T12:50:25Z', 'info': 'Looks like your service has expired. Please update your service subscription.', 'code': 4104, 'group': 4, 'retry': False}} in domain climatisation
2024-11-19 13:50:27.900 ERROR (MainThread) [homeassistant.components.home_connect.switch] An error occurred: {'description': 'HomeAppliance is offline', 'key': 'SDK.Error.HomeAppliance.Connection.Initialization.Failed'}
2024-11-19 13:50:28.676 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time (2.0 secs) to GroupValueRead request for: 11/1/1
2024-11-19 13:50:28.676 WARNING (MainThread) [xknx.log] Could not sync group address '11/1/1' (Innentemperatur - Value)
[... cut ...]
2024-11-19 13:53:42.906 WARNING (SyncWorker_8) [weconnect] /vehicles/WVGZZZE27RE004246/domains/chargingProfiles/chargingProfilesStatus/profiles/1/options: Unknown attribute usePrivateCurrentEnabled with value True
2024-11-19 13:53:42.920 WARNING (SyncWorker_8) [weconnect] /vehicles/WVGZZZE27RE004246: Unknown attribute climatisationTemperatureOutside with value {'error': {'message': 'Bad Gateway', 'errorTimeStamp': '2024-11-19T12:53:42Z', 'info': 'Looks like your service has expired. Please update your service subscription.', 'code': 4104, 'group': 4, 'retry': False}} in domain climatisation
```

marq24, 2024-11-19:
Mahlzeit... 

1) ich habe Deinen Log output massiv bearbeitet - da diverse Address, ID und eMail informationen als PlainText enthalten waren... 

2) Mein Stand ist, dass man in der Senec App (und auch im Webportal) kein "einzelnen" Speicher auswählen kann - die Daten über die Systeme aufsummiert werden und dann als EIN Wert (für das als MASTER konfigurierte System) ausgeliefert werden...

Ich kann das hier lokal natülich nicht nachstellen, da ich kein zweites Senec System habe (und die Wahrscheinlichkeit das ein weiteses System dazu kommt geht 'lim 0'... 

Also meine aktuelle info ist, die Senec App/Webportal zeigt die Daten aller Systeme zusammen an / es ist keine Einzelsystem Betrachtung möglich... So ist auch die WebAPI Integration umsetetzt.

marq24, 2024-11-24:
@be1001 hast Du hier weiteres feedback für mich? Sonst mach' doch dne issue bitte zu - Merci

be1001, 2024-11-24:
Hallo,

bin noch mit Senec dran, ob es eine Lösung gibt, oder ich 3 Logins benötige

marq24, 2024-11-24:
Nur nochmal als Info - bei den mir bislang bekannten Usern, die mehr als ein SENEC System installiert hatten/haben, wurden die Messwerte aller System auf Seiten von SENEC summiert und so getan als wäre es ein System (dieses ist dann auch als MASTER geflagged)... Die Integration ist in der Lage das Master System zu erkennen und so alle Daten zu liefern (so zumindest der Plan und die bisherigen Rückmeldungen von Usern)... 

... und viel Erfolg bei dem Versuch mit Senec sinnhaftig zu kommunizieren...

be1001, 2024-11-24:
Masterlösung ist in Ordnung wenn alle Systeme am selben Ort stehen, in meinem Fall sind die beiden P4 Systeme aber nicht am selben Ort, haben auch zwei verschiedene Zähler und PV Anlagen. Der einzige gemeinsame Nenner ist der Kunde und die Netzwerkumgebung.
Der Fehler war scheinbar bereits bei der Einrichtung den Kunden mehrfach zu benutzen und nicht einen neuen Kunden für jeden Standort anzulegen.

marq24, 2024-11-24:
Danke für die zusätzliche Erklärung - das ist dann m.E. definitiv ein SENEC Thema... Du bist Eigentümer von mehreren Immobilien mit getrennten PV Anlagen - das darf dann bei SENEC nicht unter einer 'Nummer' laufen.

Ich mach das Ticket hie zu, da die Trennung schon bei SENEC erfolgen muss und Du dann die Integration "einfach" mehrfach mit unterschiedlichen Logins verwenden kannst/sollst/musst.

closed 2024-11-24 by marq24