marq24/ha-senec-v3 #118: Data update too slow

floball44, 2025-01-10:
Good morning. I have had an E4 storage device since yesterday. Yesterday the update interval was 60 seconds. Today the values are updated irregularly and never less than 8 minutes up to 30 minutes. I set the interval in the configuration to 60 seconds, but nothing changes. Unfortunately, the data cannot be used like this. What can I do? Thanks for the help.

marq24, 2025-01-10:
check your HA logs for messages from the integration - probably your Senec simply does not provide the information (reject the request that will be send ever 60sec)...

floball44, 2025-01-10:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: homeassistant
Quelle: custom_components/senec/pysenec_ha/__init__.py:3648
Integration: SENEC.Home (Dokumentation, Probleme)
Erstmals aufgetreten: 08:25:48 (7 Vorkommnisse)
Zuletzt protokolliert: 09:00:23

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 267, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 478, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 177, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 359, in state
    return self.value
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 410, in value
    if (native_value := self.native_value) is None:
                        ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/number.py", line 90, in native_value
    value = getattr(self.coordinator.senec, self.entity_description.key)
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 3648, in spare_capacity
    return int(self._spare_capacity)
ValueError: invalid literal for int() with base 10: ''

marq24, 2025-01-10:
looks like that you have enabled the `spare_capacity` number entity while you are not subscribed to this feature in your senec contract... 

As a quick fix you simply can disable this number entity - or wait till I am going to release a fix (that will handle this error)

marq24, 2025-01-10:
please update to https://github.com/marq24/ha-senec-v3/releases/tag/2025.1.0 and close this issue when it's solved with the new version - TIA