marq24/ha-senec-v3 #120: Sensorwerte haben zu viele Stellen

electricdelicate, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
Alle Werte werden korrekt ausgegeben, bis zur xsten Stelle nach dem Komma. Das ist leider zuviel für die Custom Lovelace Card. 
https://github.com/io-debug/io-senec-card
Leider werden dort, und nur dort, die Werte trotz der heruntergestellten HA  Sensor "Precision" auf die 2. Stelle nach dem Komma, in voller länge angezeigt. Wünschenswert wären grundsätzlich nur 2 Stellen nach dem Komma.

Jetzt weiß ich nicht an wo das Problem liegt bei deinem Repository oder der der Lovelace Card oder bei mir.

Wäre schön wenn man eine Lösung dafür finden könnte. :)

So siehts aus:

[image]

[image]

marq24, 2025-01-25:
Du kannst bei jedem Sensor die Anzahl der Nachkommastellen selbst einstellen...

electricdelicate, 2025-01-25:
> Du kannst bei jedem Sensor die Anzahl der Nachkommastellen selbst einstellen...

Meinst du die Precision? Die greift hier nicht ...

[image]

marq24, 2025-01-25:
Ja die mein(te) ich - wenn die nicht greift, dann "kann" ich da sicherlich was patchen (was aber auch eigentlich diese Card selber implementieren sollte...  -> aber in der Tat ist bei meiner Senec App die Stellenanzahl "recht" (aka "zu") hoch...

electricdelicate, 2025-01-25:
> Ja die mein(te) ich - wenn die nicht greift, dann "kann" ich da sicherlich was patchen (was aber auch eigentlich diese Card selber implementieren sollte... -> aber in der Tat ist bei meiner Senec App die Stellenanzahl "recht" (aka "zu") hoch...

Das wäre super. :)

electricdelicate, 2025-01-27:
> Ja die mein(te) ich - wenn die nicht greift, dann "kann" ich da sicherlich was patchen (was aber auch eigentlich diese Card selber implementieren sollte... -> aber in der Tat ist bei meiner Senec App die Stellenanzahl "recht" (aka "zu") hoch...

Hi Marq, wollte nur nochmal fragen ob du was reinpatchst oder es doch der Entwickler der Card. Habe so das Gefühl das die Card nicht mehr weiterentwickelt wird. Habe das Problem auch dort geposted, bis jetzt aber keinerlei reaktion. Was natürlich sehr schade wäre. Beides zusammen ist mMn wirklich das perfekte Home Assistant Senec Paket.

marq24, 2025-01-27:
... gib mir ein paar Tage

marq24, 2025-01-27:
Ok - I have some feedback - looks like that this is not 'easy' fixable - since all Sensors have the corresponding attribute set - but for what ever reason HA states will be provided as 'raw'...

You can try/do the following - edit your local version of the card `io-senec.js` : 

https://github.com/io-debug/io-senec-card/blob/78acdf9c11c1f5307b12a837868d79441449381f/io-senec.js#L34C26-L34C40

and replace this line with:

```
val numOrStr = stateObj.state;
if (!isNaN(parseFloat(numOrStr)) && !isNaN(numOrStr - 0)) {
	state = String(Math.round(numOrStr * 100) / 100);
} else {
	state = String(numOrStr);
}
```
this will give you two digits...

electricdelicate, 2025-01-27:
> Ok - I have some feedback - looks like that this is not 'easy' fixable - since all Sensors have the corresponding attribute set - but for what ever reason HA states will be provided as 'raw'...
>
> You [...]

OK, first thank you for your feedback!
But I think my syntax is wrong, sorry I´m a noob in coding, the change in the file have no effect so far.

[image]

marq24, 2025-01-27:
Any code change, always required a restart of HA - but I assume @io-debug will sooner or later merge the PR https://github.com/io-debug/io-senec-card/pull/2 (or fix it correctly in the sources)... He have also provided multiple changes to this integration here - I am confident he will reply - for what ever reason people also have a RL ?! :-D

ETH1166, 2025-01-27:
Hello Matthias @marq24,

as I always rely on your coding and support I had also tried your patch manually:

> if (stateObj) {
>           // Original Code: state = String(stateObj.state);
>           // Replaced by the following 6 lines to round numbers to 2 digits precision
>             val numOrStr = stateObj.state;
>             if (!isNaN(parseFloat(numOrStr)) && !isNaN(numOrStr - 0)) {
> 	            state = String(Math.round(numOrStr * 100) / 100);
>             } else {
> 	            state = String(numOrStr);
>             }
>         }
>       data[i] = state;

I just had directly edited the original file `/config/www/community/io-senec-card/io-senec.js`.

Even after a restart of HomeAssistant, the number of digits is not reduced to 2, but stays the same. I am lacking skills of debugging HASS. Might there be an issue with the if() clause and the isNaN() logic and we always fall back to the else-clause? (I am also lacking JS skills...)

Thank you for your support!
Best regards
Thomas

ETH1166, 2025-01-27:
Hello @marq24,
I just have seen your PR on io-debug, which is using `if (i > 0) {` instead. I had also tried this, but the result is the same: I cannot get the digits removed, even after restarting HomeAssistant and clearing my browser cache...
Best regards
Thomas

P.S.: Sorry, I did not mean to bother you...

io-debug, 2025-01-27:
> Any code change, always required a restart of HA - but I assume [@io-debug](https://github.com/io-debug) will sooner or later merge the PR [io-debug/io-senec-card#2](https://github.com/io-debug/io-senec-card/pull/2) (or fix it correctly in the sources)... He have also provided multiple changes to this integration here - I am confident he will reply - for what ever reason people also have a RL ?! :-D

Hey, I've merged your PR. I currently lack the time to respond quickly. :)

electricdelicate, 2025-01-27:
> Hey, I've merged your PR. I currently lack the time to respond quickly. :)

Great, thank you so much!

[image]

marq24, 2025-01-27:
So with the change in the senec-card all is solved, then I close this...
@ETH1166 - I hope the updated card will also work for you

I guess the root cause is, that you must flush local cache (so that new js get's requested)...

marq24, 2025-01-27:
> Hey, I've merged your PR. I currently lack the time to respond quickly. :)

This is all just a hobby for all of us - so nobody can expect that there is a reaction from the developer in < 48h... I just don't have a real life so I might be a exception ;-)

closed 2025-01-27 by marq24