marq24/ha-senec-v3 #122: Wallbox mode is properly pushed to the webapi, but the entity in HomeAssistant never gets a valid value (defaults to UNKNOWN)

mschmiedel, 2025-02-02:
### With which version of the integration you have an issue?

WEB.API: mein-senec.de Portal

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I guess this must be a mapping issue at some point. I use both the V3Hybrid.... and WebApi integrations. When I set a different Wallbox mode, it is properly being mapped to the webapi. However, the entity as such in HomeAssistant stores no value, it always resets to "Unknown". 

My first guess was to add a debug line to wallbox_1_mode method in the pysenec_ha and test it, but this code is never being called. The chargingMode value seems to be correct for both the local and the webApi, so I guess there must be an issue with the select config, as other field types seem to be storing values properly... 

### Add your DEBUG log output

```shell
homeassistant  | 2025-02-02 10:07:16.887 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API app_update_wallboxes for '1' wallboxes
homeassistant  | 2025-02-02 10:07:16.887 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_wallbox_data(self) ********
homeassistant  | 2025-02-02 10:07:16.887 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_data(self) ********
homeassistant  | 2025-02-02 10:07:16.887 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API get https://app-gateway-prod.senecops.com/v1/senec/anlagen/XXXXX/wallboxes/1
homeassistant  | 2025-02-02 10:07:17.276 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API response: {'id': 1, 'configurable': True, 'maxPossibleChargingCurrentInA': 16.0, 'minPossibleChargingCurrentInA': 6.0, 'chargingMode': 'SMART_SELF_GENERATED_COMPATIBILITY_MODE', 'currentApparentChargingPowerInVa': 0.0, 'electricVehicleConnected': False, 'hasError': False, 'statusText': 'not connected', 'configuredMaxChargingCurrentInA': 16.0, 'configuredMinChargingCurrentInA': 6.0, 'temperatureInCelsius': 7.9, 'numberOfElectricPowerPhasesUsed': 0, 'allowIntercharge': False, 'compatibilityMode': True, 'wallboxChargingPowerStats': {'onePhase': {'min': 1.3800000000000001, 'max': 3.68}, 'twoPhase': {'min': 2.7600000000000002, 'max': 7.36}, 'threePhase': {'min': 4.140000000000001, 'max': 11.040000000000001}}}
homeassistant  | 2025-02-02 10:07:17.277 DEBUG (MainThread) [custom_components.senec] read: {'now': {'powerGenerationInW': 2758.677905758485, 'powerConsumptionInW': 835.16, 'gridFeedInInW': 11.754825415200745, 'gridDrawInW': 20.781258084126655, 'batteryChargeInW': 1932.5205821864652, 'batteryDischargeInW': 1e-05, 'batteryLevelInPercent': 5.0, 'selfSufficiencyInPercent': 97.51, 'wallboxInW': 1e-05}, 'today': {'powerGenerationInWh': 1747.4174499511719, 'powerConsumptionInWh': 6718.39, 'gridFeedInInWh': 52.36554145812988, 'gridDrawInWh': 4768.795013427734, 'batteryChargeInWh': 760.6163024902344, 'batteryDischargeInWh': 1015.9740447998047, 'batteryLevelInPercent': 5.0, 'selfSufficiencyInPercent': 29.02, 'wallboxInWh': 0.0}, 'total': {'totals': None, 'measurements': ['POWER_GENERATION', 'POWER_CONSUMPTION', 'GRID_IMPORT', 'GRID_EXPORT', 'BATTERY_IMPORT', 'BATTERY_EXPORT', 'BATTERY_LEVEL_IN_PERCENT', 'AUTARKY_IN_PERCENT', 'WALLBOX_CONSUMPTION'], 'timeseries': [{'date': 1735686000, 'measurements': {'durationInSeconds': 5220435, 'values': [60.858131408691406, 102.23071618652344, 45.24259948730469, 3.4476828575134277, 20.9287052154541, 21.769269943237305, 8.155978202819824, 42.53, 23.506]}}]}, 'total_old': None, 'tech_data': {'systemOverview': {'systemId': 321509, 'hid': None, 'productName': 'SENEC.HOME V3 hybrid LFP', 'installationDate': 0}, 'casing': {'serial': 'XXXXX', 'temperatureInCelsius': 24.817676544189453}, 'mcu': {'mainControllerSerial': 'XXXX', 'mainControllerState': {'name': 'LADEN', 'severity': 'INFO'}, 'firmwareVersion': '830', 'guiVersion': 969}, 'warranty': {'endDate': 0, 'warrantyTermInMonths': 0}, 'batteryModules': [], 'batteryInverter': {'state': {'name': None, 'severity': None}, 'vendor': None, 'firmware': {'firmwareVersion': None, 'firmwareVersionHumanMachineInterface': None, 'firmwareVersionPowerUnit': None, 'firmwareVersionBidirectionalDcConverter': None}, 'temperatures': {'amb': None, 'halfBridge1': None, 'halfBridge2': None, 'throttle': None, 'max': None}, 'lastContact': {'time': 61675682400000, 'timestamp': '3924-06-04T22:00:00Z', 'severity': 'ERROR'}, 'flags': []}, 'batteryPack': {'numberOfBatteryModules': 0, 'technology': 'UNKNOWN', 'maxCapacityInKwh': 0.0, 'maxChargingPowerInKw': None, 'maxDischargingPowerInKw': None, 'currentChargingLevelInPercent': 0.0, 'currentVoltageInV': 0.0, 'currentCurrentInA': 0.0, 'remainingCapacityInPercent': 0.0}, 'installer': {}, 'wallbox': [{'id': 1, 'configurable': True, 'maxPossibleChargingCurrentInA': 16.0, 'minPossibleChargingCurrentInA': 6.0, 'chargingMode': 'SMART_SELF_GENERATED_COMPATIBILITY_MODE', 'currentApparentChargingPowerInVa': 0.0, 'electricVehicleConnected': False, 'hasError': False, 'statusText': 'not connected', 'configuredMaxChargingCurrentInA': 16.0, 'configuredMinChargingCurrentInA': 6.0, 'temperatureInCelsius': 7.9, 'numberOfElectricPowerPhasesUsed': 0, 'allowIntercharge': False, 'compatibilityMode': True, 'wallboxChargingPowerStats': {'onePhase': {'min': 1.3800000000000001, 'max': 3.68}, 'twoPhase': {'min': 2.7600000000000002, 'max': 7.36}, 'threePhase': {'min': 4.140000000000001, 'max': 11.040000000000001}}}, None, None, None]}
```

marq24, 2025-02-02:
WTF! `SMART_SELF_GENERATED_COMPATIBILITY_MODE`

marq24, 2025-02-02:
... do you have also the LOCAL SENEC Integration configured?! [and can you please be so kind and provide also the log output from there]? -> the WB status will be written via WEB -but then READ from LOCAL...

mschmiedel, 2025-02-02:
It was not logging much more, the last line was from the local senec integration. I just greped everything that has "custom_components.senec" from the logs... 

I See stuff like this: 
`homeassistant  | 2025-02-02 10:13:26.517 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.278 seconds (success: True)
homeassistant  | 2025-02-02 10:13:56.239 DEBUG (MainThread) [custom_components.senec] _async_update_data called`

but not much more. It feels as if the select is not trying to call any refresh state after a change was done via the UI...

marq24, 2025-02-02:
let me ask different - how many Senec integrations you have configured? Based on the provided debug log output it looks like, that you have (only) installed the WebAPI Integration? Looks like there is no output from the local/lala.cgi Integration... If this is the case please note, that the Wallbox support requires that there are WebAPI-Base + local'lala.cgi'-Based must be configured.

mschmiedel, 2025-02-02:
100% sure both:

[image]

marq24, 2025-02-02:
mhhh - I really don't understand than, that you do not have any log output from the local based integration... -> something like this `... DEBUG (MainThread) [custom_components.senec] read: {'data': ...` so `data` is the keyword here...

marq24, 2025-02-03:
ok - let me try to rephrase it - from the so far shared log output it looks like that there is some sort of miss-configuration/malfunction during system startup - at least I do not have an explanation so far. I would kindly ask you to:
 
1. Update your cloned repo to the lastest changes I have commit to master 
2. Install the version from your (update) master branch
3. enable the debug log as described in the integration docs
4. restart HA
5. provide me with the full log file generated by HA during the initialization phase of the integration (both configurations must log different messages) - use eMail for that

TIA

mschmiedel, 2025-02-05:
OK, I will check it, but will not have time for it until the weekend... ;)

marq24, 2025-02-06:
take your time - I am not in a rush...

mschmiedel, 2025-02-07:
Hi, I sent you the log via email, as it is rather long and I was lazy to mask out sensitive data... ;)

mschmiedel, 2025-02-07:
Here a minor check I could do:

[image]

causes this log:
homeassistant  | 2025-02-07 17:32:16.536 DEBUG (MainThread) [custom_components.senec] Start value update: optimized_4

mschmiedel, 2025-02-07:
I guess the issue is in this method:

[image]

look at the logs:
homeassistant  | 2025-02-07 18:08:29.933 DEBUG (MainThread) [custom_components.senec.select] Value from else: optimized_3
homeassistant  | 2025-02-07 18:08:29.934 DEBUG (MainThread) [custom_components.senec.select] Value from previous else: None
homeassistant  | 2025-02-07 18:08:29.934 DEBUG (MainThread) [custom_components.senec.select] Value from unknown: unknown

the second if-else block is resetting the value, the first one was the correct one and it gets overwritten with an empty value...

marq24, 2025-02-07:
thanks for your investigation - and patience this `else` really does not make sense...

OLD:
```
            if (value is None or str(value) == LOCAL_WB_MODE_UNKNOWN) and self._previous_value is not None:
                value = self._previous_value
            else:
                value = None
```

NEW:
```
            if (value is None or str(value) == LOCAL_WB_MODE_UNKNOWN) and self._previous_value is not None:
                value = self._previous_value
```

so when you altered the code like this - I guess we have finally the correct value?

mschmiedel, 2025-02-07:
Yep, first try without altering the value:

[image]

After altering a few times:
[image]

This is great, now I can use the value of the select, I was waiting for that, I need it in a few scripts. Thanks 😉

mschmiedel, 2025-02-07:
One more minor issue with the wording...

OPTIMIERT (Ladeunterbrechung EIN) means that it will load with at least the icmin set.
OPTIMIERT (Ladeunterbrechung AUS) means it will shut down if PV energy is not enough. 

In the app it is marked as "Ladeunterbrechung verhindern". This feels a bit odd in HA. Maybe a better (shorter) wording can be found that will cover the intention of the settings better?

marq24, 2025-02-07:
OPTIMIERT (Ladeunterbrechung EIN) means that it will load with at least the icmin set.
OPTIMIERT (Ladeunterbrechung AUS) means it will shut down if PV energy is not enough.

My understanding is just the other way around  - `Ladeunterbrechung EIN` means (IMHO) that's possible that the charging can be interrupted (for what ever reason)...

and (JFYI) I will rename `OPTIMIERT (1/Ladeunterbrechung EIN)` and `OPTIMIERT (0/Ladeunterbrechung AUS)`

marq24, 2025-02-07:
I hope you can close this with https://github.com/marq24/ha-senec-v3/releases/tag/2025.2.0

mschmiedel, 2025-02-07:
Yep, feels fine now. About wording and meaning of the values, let's get the opinion of people with V2.1/V3 and 4. gen devices, just to be sure there is no misunderstanding. Maybe it is device dependent.

closed 2025-02-07 by mschmiedel