marq24/ha-senec-v3 #123: Add senec_wallbox_1_set_icmin

mschmiedel, 2025-02-02:
**Is your feature request related to a problem? Please describe.**
From my perception, the field senec_wallbox_1_set_icmax cannot be set, even though it is in the api. The value that the Senec App uses for the WB Current is senec_wallbox_1_set_icmin.

**Describe the solution you'd like**
Add senec_wallbox_1_set_icmin to be configureable just like icmax, this should primarily be pushed to the webapi rather than the local instance.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

Posts like this always cause a response looking like this:

[image]

So from my perspective, icmax settings have no impact on the webapi. However, setting the icmax locally (without the web integration) will allow the wallbox to charge with the given current until the storage makes a new refresh with the server. ;)

marq24, 2025-02-02:
@iot-sle can you remember, why we did not implement the SET_ICMIN ? [I can't remember the details to be honest] It was IMHo related to the value of the attr `configuredMinChargingCurrentInA` (since we always set the IC_MIN to this value ?!

mschmiedel, 2025-02-07:
Just as an extra comment, the behavior I am reporting is valid for v2.1 and v3 Systems, maybe the newer 4/E4/P4 devices behave differently...

iot-sle, 2025-02-19:
as my WB has a maximum charging current of 32 A. I tried first to use limit it by max charging current. The result: It was always changed during the next refresh. The idea was to limit the the charging current by using the "solar optimized" feature and setting the IC_Min  to that value, which should be the maximum charging current. By example setting IC_Min to 10A and no solar power available it results in 10A charging current. So it's an workaroung to limit the charging current, because the ic max is not working as expected.

mschmiedel, 2025-02-19:
icMax does work, I use it via rest command, but senec refreshes it every 5 minutes the WB communicates with the servers, but since I set it every 30 seconds, it is maneagable.

However, I will do further testing when I have some time...

mschmiedel, 2025-02-23:
Well, I have done some more tests on my side...

1. The icmax is not working for me at all. From my experience, the rest api will never take this value, as this is calculated by senec on server side, so it seems to be read-only. It only makes sense to send this value to the local senec device. As @iot-sle said, this will be overwritten eventually, but for solar tracking I write it every 30 seconds, this gets me a decent behavior. 

2. The icmin has the sole purpose of limiting the Wallbox to a minimum current to use. This can be interesting for tuning the priority of the wallbox in relation to the battery: a higher icmin might give less power to the battery... This can and should be set by the api and is reflected in the app. It is also the minimum charging current if "Ladeunterbrechung verhindern" is turned on. 

This is the sample macro how I write the ic_max to the device with a rest_command. the sensor senec_wb_target_current I use is a template of my own that has gotten quite complex in order to cover everything I need... 😉 In case the values are properly encoded in ieee_754, I am not sure what is going on then, I have not had a chance to check the code.

`  wb_current:
    url: https://npu/lala.cgi
    method: POST
    payload: >
            {% macro float_ieee754_single(value) -%}
            {% set dl = pack(value|float, "!f").decode('latin-1')|map('ord')|list -%}
            {{ ("{:02x}"*dl|length).format(*dl) -}}
            {% endmacro -%}
            {% set targetCurrent = states('sensor.senec_wb_target_current')|float %}
            {% set targetCurrent = 0|float if targetCurrent < 6 else targetCurrent %}
            {% set hexCurrent = float_ieee754_single(targetCurrent) %}
            { "WALLBOX" : {"SET_ICMAX": ["fl_{{hexCurrent}},“fl_00000000","fl_00000000","fl_00000000"]} }
    verify_ssl: false`

So my expected behavior would be:

1. Have the slider for icmin that only reflects to the api
2. Allow setting icmax, but no need of a slider or nice UI interface, just send the ieee_754 value to lala.cgi. This should only be used in automations (instead of my rest command).

@iot-sle, what are your thoughts about this? Since I have a v3 system, it would be important to make sure this does not break anything on v4 systems, where icmax would end up being irrelevant as there is no local interface afaik...

marq24, 2025-08-14:
is that still an issue with the latest Integration Version? -> IMHO icmax was always some sort of icmin... at least code wise...

closed 2025-09-07 by marq24