marq24/ha-senec-v3 #127: Add current value of "spare capacy"

iot-sle, 2025-02-16:
The current value of "spare capacy" ("Notstromreserve") could be easily added to the values of batteries. 
It can be found in lala.cgi section "BAT1" variable "SPARE_CAPACITY" an number between 0-100.

iot-sle, 2025-02-19:
It's fixed now. Setting is only possible by webapi. But since change of batteries to LFP, the setting is NA in HA SenecWebApi any longer. I guess the name of property had been changed by SENEC. If I find the new name I will keep you informed. Thank you!

closed 2025-02-19 by iot-sle

marq24, 2025-02-19:
... I still can "set" the value via my web-api integration - but the display value is not correct ... so there is for sure some "senic-magic"...

@iot-sle  You you have also feedback for me #123 ?!