marq24/ha-senec-v3 #128: SENEC WebAPI cant add/reconnect to

indikat0r, 2025-03-08:
### With which version of the integration you have an issue?

WEB.API: mein-senec.de Portal

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hello,
my WebAPI intgeration stopped working.
I have removed it, and now i cant it readd.
i have added the Debug Log.
as far as i understand, the url returns empty data?

### Add your DEBUG log output

```shell
2025-03-08 11:08:33.674 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: 'pv@maier-wang.de' without options
2025-03-08 11:08:33.674 DEBUG (MainThread) [custom_components.senec.pysenec_ha] MySenecWebPortal initialized and IntBridge.app_api set to <custom_components.senec.pysenec_ha.MySenecWebPortal object at 0x7fd04647ae90>
2025-03-08 11:08:33.674 DEBUG (MainThread) [custom_components.senec.pysenec_ha] IntBridge.avail() -> IntBridge.app_api: True IntBridge.lala_cgi: True
2025-03-08 11:08:33.674 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_authenticate(self) ********
2025-03-08 11:08:34.045 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API: response:{'token': 'cHZAbWFpZXItd2FuZy5kZTpzVExydHg1RA==', 'refreshToken': ''}
2025-03-08 11:08:34.045 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: get_master_plant_id(self) ********
2025-03-08 11:08:34.301 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API response: [{'id': '253699', 'steuereinheitnummer': 'S36700801093881874540162098', 'gehaeusenummer': 'DE-V3-H-10LI10-14556', 'strasse': 'Am Kirchfeld', 'hausnummer': '46 B', 'postleitzahl': '85368', 'ort': 'Volkmannsdorf', 'zeitzone': 'Europe/Berlin', 'wallboxIds': [], 'systemType': 'V3'}]
2025-03-08 11:08:34.301 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API set _app_master_plant_id to 253699
2025-03-08 11:08:34.301 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API set _app_wallbox_num_max to 0
2025-03-08 11:08:34.301 INFO (MainThread) [custom_components.senec.pysenec_ha] APP-API: Login successful
2025-03-08 11:08:34.302 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** app_update_context(self) ********
2025-03-08 11:08:34.302 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_tech_data(self) ********
2025-03-08 11:08:34.302 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_data(self) ********
2025-03-08 11:08:34.302 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API get https://app-gateway.prod.senec.dev/v1/senec/systems/253699/technical-data
2025-03-08 11:08:36.999 WARNING (MainThread) [homeassistant.helpers.entity] Update of calendar.ms365_calendar_geburtstagskalender is taking over 10 seconds
2025-03-08 11:08:42.863 DEBUG (MainThread) [custom_components.senec] _async_update_data called
2025-03-08 11:08:43.136 DEBUG (MainThread) [custom_components.senec.pysenec_ha] got cookie update: senec-homev3-gui=81a3c36368d8e536882a7037612171aa0d09c2957cee590c1c535c809f196c0b;Secure;HttpOnly;Max-Age=3600;SameSite=Strict
2025-03-08 11:08:43.140 DEBUG (MainThread) [custom_components.senec] read: {'data': {'TEMPMEASURE': {'BATTERY_TEMP': 33.900001525878906, 'CASE_TEMP': 33.82106399536133, 'MCU_TEMP': 50.28357696533203}, 'PV1': {'POWER_RATIO': 100.0, 'POWER_RATIO_L1': 100.0, 'POWER_RATIO_L2': 100.0, 'POWER_RATIO_L3': 100.0, 'MPP_VOL': [215.7360076904297, 247.05201721191406, 0.0], 'MPP_CUR': [6.319000244140625, 0.0, 0.0], 'MPP_POWER': [1358.509033203125, 0.0, 0.0]}, 'PWR_UNIT': {'POWER_L1': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L2': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L3': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]}, 'PM1OBJ1': {'FREQ': 50.06999969482422, 'U_AC': [229.5, 234.40000915527344, 231.8000030517578], 'I_AC': [5.779999732971191, 11.069999694824219, 12.59000015258789], 'P_AC': [-1274.18994140625, -2556.550048828125, -2733.06982421875], 'P_TOTAL': -6563.81982421875}, 'PM1OBJ2': {'FREQ': 50.0099983215332, 'U_AC': [230.10000610351562, 234.60000610351562, 232.0], 'I_AC': [7.199999809265137, 7.139999866485596, 7.179999828338623], 'P_AC': [-1653.8900146484375, -1672.010009765625, -1662.2799072265625], 'P_TOTAL': -4988.1796875}, 'ENERGY': {'CAPTESTMODULE': [2839.404052734375, 2837.527099609375, 2847.83349609375, 2850.022705078125], 'GRID_POWER_OFFSET': 'VARIABLE_NOT_FOUND', 'GUI_BAT_DATA_COLLECTED': 1, 'GUI_BAT_DATA_CURRENT': 46.400001525878906, 'GUI_BAT_DATA_FUEL_CHARGE': 59.0, 'GUI_BAT_DATA_MAX_CELL_VOLTAGE': 3363, 'GUI_BAT_DATA_MIN_CELL_VOLTAGE': 3355, 'GUI_BAT_DATA_POWER': 2491.68017578125, 'GUI_BAT_DATA_VOLTAGE': 53.70000076293945, 'GUI_BOOSTING_INFO': 0, 'GUI_CAP_TEST_DIS_COUNT': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_START': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_STATE': 0, 'GUI_CAP_TEST_STOP': 'VARIABLE_NOT_FOUND', 'GUI_CHARGING_INFO': 1, 'GUI_FACTORY_TEST_FAN': 'VARIABLE_NOT_FOUND', 'GUI_GRID_POW': -6563.81982421875, 'GUI_HOUSE_POW': 0.0, 'GUI_INIT_CHARGE_START': 0, 'GUI_INIT_CHARGE_STOP': 0, 'GUI_INVERTER_POWER': 6346.6884765625, 'GUI_TEST_CHARGE_STAT': 0, 'GUI_TEST_DISCHARGE_STAT': 0, 'INIT_CHARGE_ACK': 0, 'INIT_CHARGE_DIFF_VOLTAGE': 0.0, 'INIT_CHARGE_MAX_CURRENT': 0.0, 'INIT_CHARGE_MAX_VOLTAGE': 0.0, 'INIT_CHARGE_MIN_VOLTAGE': 0.0, 'INIT_CHARGE_RERUN': 0, 'INIT_CHARGE_RUNNING': 0, 'INIT_CHARGE_STATE': 0, 'INIT_CHARGE_TIMER': 0, 'INIT_DISCHARGE_MAX_CURRENT': 0.0, 'LI_STORAGE_MODE_RUNNING': 0, 'LI_STORAGE_MODE_START': 0, 'LI_STORAGE_MODE_STOP': 0, 'SAFE_CHARGE_FORCE': 0, 'SAFE_CHARGE_PROHIBIT': 0, 'SAFE_CHARGE_RUNNING': 0, 'STAT_HOURS_OF_OPERATION': 19626, 'STAT_LIMITED_NET_SKEW': 0, 'STAT_STATE': 14, 'TEST_CHARGE_ENABLE': 'VARIABLE_NOT_FOUND', 'TEST_CYCLE': 'VARIABLE_NOT_FOUND', 'ZERO_EXPORT': 0}, 'LOG': {'USER_LEVEL': 'VARIABLE_NOT_FOUND', 'LOG_IN_NOK_COUNT': 0}, 'BAT1': {'SPARE_CAPACITY': 0}, 'STATISTIC': {}, 'SOCKETS': {'ALREADY_SWITCHED': [0, 0], 'ENABLE': [0, 0], 'FORCE_ON': [0, 0], 'LOWER_LIMIT': [0, 0], 'NUMBER_OF_SOCKETS': 2, 'POWER_ON': [0, 0], 'POWER_ON_TIME': [0, 0], 'PRIORITY': [0, 0], 'RESET_SWITCHED': 0, 'SWITCH_ON_HOUR': [0, 0], 'SWITCH_ON_MINUTE': [0, 0], 'TIME_LIMIT': [0, 0], 'TIME_REM': [0, 0], 'UPPER_LIMIT': [0, 0], 'USE_TIME': [0, 0]}, 'BMS': {'CELL_TEMPERATURES_MODULE_A': [29, 31, 33], 'CELL_TEMPERATURES_MODULE_B': [28, 30, 33], 'CELL_TEMPERATURES_MODULE_C': [0, 0, 0], 'CELL_TEMPERATURES_MODULE_D': [0, 0, 0], 'CELL_VOLTAGES_MODULE_A': [3360, 3360, 3360, 3361, 3360, 3359, 3358, 3357, 3360, 3358, 3358, 3356, 3355, 3356, 3356, 3357], 'CELL_VOLTAGES_MODULE_B': [3362, 3361, 3363, 3363, 3361, 3362, 3360, 3361, 3360, 3358, 3359, 3357, 3358, 3358, 3357, 3357], 'CELL_VOLTAGES_MODULE_C': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'CELL_VOLTAGES_MODULE_D': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'CURRENT': [23.0, 23.399999618530273], 'VOLTAGE': [53.70000076293945, 53.70000076293945], 'SOC': [59, 58], 'SOH': [100, 100], 'CYCLES': [68, 67], 'MODULES_CONFIGURED': 2}}, 'version': {'FACTORY': {'SYS_TYPE': 20, 'COUNTRY': 0, 'DEVICE_ID': '<MASKED>'}, 'WIZARD': {'APPLICATION_VERSION': '0833', 'FIRMWARE_VERSION': '2307', 'INTERFACE_VERSION': '0970'}, 'BAT1': {'TYPE': 4}, 'SYS_UPDATE': {'NPU_VER': 3, 'NPU_IMAGE_VERSION': 2408}, 'STATISTIC': {}}}
2025-03-08 11:08:43.140 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.277 seconds (success: True)
[... cut ...] 3945], 'SOC': [60, 58], 'SOH': [100, 100], 'CYCLES': [68, 67], 'MODULES_CONFIGURED': 2}}, 'version': {'FACTORY': {'SYS_TYPE': 20, 'COUNTRY': 0, 'DEVICE_ID': '<MASKED>'}, 'WIZARD': {'APPLICATION_VERSION': '0833', 'FIRMWARE_VERSION': '2307', 'INTERFACE_VERSION': '0970'}, 'BAT1': {'TYPE': 4}, 'SYS_UPDATE': {'NPU_VER': 3, 'NPU_IMAGE_VERSION': 2408}, 'STATISTIC': {}}}
2025-03-08 11:10:13.142 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.277 seconds (success: True)
```

marq24, 2025-03-08:
a bit tricky from the log output to tell what's going wrong (since local access is also running) - nevertheless - please update to the new release...

indikat0r, 2025-03-09:
`2025-03-09 08:56:29.997 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.275 seconds (success: True)
2025-03-09 08:56:38.368 WARNING (MainThread) [custom_components.senec.pysenec_ha] APP-API: NO RES - Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/253699/measurements?resolution=YEAR&from=1514764800&to=1744181767 causing: Connection timeout to host https://app-gateway.prod.senec.dev/v2/senec/systems/253699/measurements?resolution=YEAR&from=1514764800&to=1744181767
2025-03-09 08:56:38.368 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_tech_data(self) ********
2025-03-09 08:56:38.368 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_data(self) ********
2025-03-09 08:56:38.369 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API get https://app-gateway.prod.senec.dev/v1/senec/systems/253699/technical-data
2025-03-09 08:56:59.722 DEBUG (MainThread) [custom_components.senec] _async_update_data called
2025-03-09 08:56:59.993 DEBUG (MainThread) [custom_components.senec.pysenec_ha] got cookie update: senec-homev3-gui=7f410d16d76e0ee8782fd17d44eda9223c4bc7b31e7600c06a34b6aae23f05b9;Secure;HttpOnly;Max-Age=3600;SameSite=Strict
2025-03-09 08:56:59.997 DEBUG (MainThread) [custom_components.senec] read: {'data': {'TEMPMEASURE': {'BATTERY_TEMP': 29.700000762939453, 'CASE_TEMP': 33.65692138671875, 'MCU_TEMP': 50.14301300048828}, 'PV1': {'POWER_RATIO': 100.0, 'POWER_RATIO_L1': 100.0, 'POWER_RATIO_L2': 100.0, 'POWER_RATIO_L3': 100.0, 'MPP_VOL': [217.5130157470703, 245.52101135253906, 0.0], 'MPP_CUR': [6.466000080108643, 0.0, 0.0], 'MPP_POWER': [1418.9810791015625, 0.0, 0.0]}, 'PWR_UNIT': {'POWER_L1': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L2': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L3': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]}, 'PM1OBJ1': {'FREQ': 50.04999923706055, 'U_AC': [229.90000915527344, 232.3000030517578, 229.5], 'I_AC': [1.709999918937683, 5.699999809265137, 3.819999933242798], 'P_AC': [-197.01998901367188, -1236.179931640625, -392.94000244140625], 'P_TOTAL': -1826.14990234375}, 'PM1OBJ2': {'FREQ': 50.0, 'U_AC': [230.10000610351562, 232.5, 229.8000030517578], 'I_AC': [3.990000009536743, 4.019999980926514, 4.0], 'P_AC': [-913.8299560546875, -931.1699829101562, -912.7899780273438], 'P_TOTAL': -2757.800048828125}, 'ENERGY': {'CAPTESTMODULE': [2839.404052734375, 2837.527099609375, 2847.83349609375, 2850.022705078125], 'GRID_POWER_OFFSET': 'VARIABLE_NOT_FOUND', 'GUI_BAT_DATA_COLLECTED': 1, 'GUI_BAT_DATA_CURRENT': 47.0, 'GUI_BAT_DATA_FUEL_CHARGE': 16.0, 'GUI_BAT_DATA_MAX_CELL_VOLTAGE': 3311, 'GUI_BAT_DATA_MIN_CELL_VOLTAGE': 3302, 'GUI_BAT_DATA_POWER': 2486.300048828125, 'GUI_BAT_DATA_VOLTAGE': 52.900001525878906, 'GUI_BOOSTING_INFO': 0, 'GUI_CAP_TEST_DIS_COUNT': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_START': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_STATE': 0, 'GUI_CAP_TEST_STOP': 'VARIABLE_NOT_FOUND', 'GUI_CHARGING_INFO': 1, 'GUI_FACTORY_TEST_FAN': 'VARIABLE_NOT_FOUND', 'GUI_GRID_POW': -1826.14990234375, 'GUI_HOUSE_POW': 0.0, 'GUI_INIT_CHARGE_START': 0, 'GUI_INIT_CHARGE_STOP': 0, 'GUI_INVERTER_POWER': 4176.78125, 'GUI_TEST_CHARGE_STAT': 0, 'GUI_TEST_DISCHARGE_STAT': 0, 'INIT_CHARGE_ACK': 0, 'INIT_CHARGE_DIFF_VOLTAGE': 0.0, 'INIT_CHARGE_MAX_CURRENT': 0.0, 'INIT_CHARGE_MAX_VOLTAGE': 0.0, 'INIT_CHARGE_MIN_VOLTAGE': 0.0, 'INIT_CHARGE_RERUN': 0, 'INIT_CHARGE_RUNNING': 0, 'INIT_CHARGE_STATE': 0, 'INIT_CHARGE_TIMER': 0, 'INIT_DISCHARGE_MAX_CURRENT': 0.0, 'LI_STORAGE_MODE_RUNNING': 0, 'LI_STORAGE_MODE_START': 0, 'LI_STORAGE_MODE_STOP': 0, 'SAFE_CHARGE_FORCE': 0, 'SAFE_CHARGE_PROHIBIT': 0, 'SAFE_CHARGE_RUNNING': 0, 'STAT_HOURS_OF_OPERATION': 19647, 'STAT_LIMITED_NET_SKEW': 0, 'STAT_STATE': 14, 'TEST_CHARGE_ENABLE': 'VARIABLE_NOT_FOUND', 'TEST_CYCLE': 'VARIABLE_NOT_FOUND', 'ZERO_EXPORT': 0}, 'LOG': {'USER_LEVEL': 'VARIABLE_NOT_FOUND', 'LOG_IN_NOK_COUNT': 0}, 'BAT1': {'SPARE_CAPACITY': 0}, 'STATISTIC': {}, 'SOCKETS': {'ALREADY_SWITCHED': [0, 0], 'ENABLE': [0, 0], 'FORCE_ON': [0, 0], 'LOWER_LIMIT': [0, 0], 'NUMBER_OF_SOCKETS': 2, 'POWER_ON': [0, 0], 'POWER_ON_TIME': [0, 0], 'PRIORITY': [0, 0], 'RESET_SWITCHED': 0, 'SWITCH_ON_HOUR': [0, 0], 'SWITCH_ON_MINUTE': [0, 0], 'TIME_LIMIT': [0, 0], 'TIME_REM': [0, 0], 'UPPER_LIMIT': [0, 0], 'USE_TIME': [0, 0]}, 'BMS': {'CELL_TEMPERATURES_MODULE_A': [26, 27, 29], 'CELL_TEMPERATURES_MODULE_B': [25, 26, 29], 'CELL_TEMPERATURES_MODULE_C': [0, 0, 0], 'CELL_TEMPERATURES_MODULE_D': [0, 0, 0], 'CELL_VOLTAGES_MODULE_A': [3309, 3308, 3309, 3308, 3307, 3307, 3307, 3306, 3307, 3306, 3306, 3305, 3304, 3304, 3304, 3302], 'CELL_VOLTAGES_MODULE_B': [3309, 3309, 3311, 3310, 3310, 3310, 3308, 3309, 3309, 3306, 3307, 3306, 3307, 3307, 3305, 3304], 'CELL_VOLTAGES_MODULE_C': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'CELL_VOLTAGES_MODULE_D': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'CURRENT': [23.600000381469727, 23.399999618530273], 'VOLTAGE': [52.79999923706055, 52.900001525878906], 'SOC': [16, 16], 'SOH': [100, 100], 'CYCLES': [69, 67], 'MODULES_CONFIGURED': 2}}, 'version': {'FACTORY': {'SYS_TYPE': 20, 'COUNTRY': 0, 'DEVICE_ID': '<MASKED>'}, 'WIZARD': {'APPLICATION_VERSION': '0833', 'FIRMWARE_VERSION': '2307', 'INTERFACE_VERSION': '0970'}, 'BAT1': {'TYPE': 4}, 'SYS_UPDATE': {'NPU_VER': 3, 'NPU_IMAGE_VERSION': 2408}, 'STATISTIC': {}}}
2025-03-09 08:56:59.997 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.275 seconds (success: True)
[... cut ...]
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 1193, in create_connection
    transport, protocol = await self._create_connection_transport(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
        ssl_shutdown_timeout=ssl_shutdown_timeout)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 1226, in _create_connection_transport
    await waiter
asyncio.exceptions.CancelledError
`

Done, but still =/

marq24, 2025-03-09:
What I don't understand is, that the APP-API requests run into an timeout - looks like that the master_plant_id that is received during the initial setup is not the correct one [253699] - can you check in the webportal what's listed there as your number?

And if it's different - then please: a) let me know the 'correct' one & b) remove the web-api part again - enable debug logging of the integration - and then add the web-api part again - and send me (by mail) the complete log output - I hope there are traces from the correct number... 

Also something you might like to check, if there are no internal firewall's DNS-blockers running in your network - I assume the default Senec (MobilApp) still works fine?

indikat0r, 2025-03-09:
Where and how can i find my master_plant_id?
is it: "Seriennummer Steuerung" or "Gehäusenummer"?

marq24, 2025-03-09:
I am sorry - just checked myself, if I can find my master_plant_id on the mein-senec.de website - or in the requests/responses - but had no luck :-/ -> looks like it's time to use Bruno (aka Postman) and debug your account...

indikat0r, 2025-03-09:
Installed Bruno and found that on google:

https://www.postman.com/blue-moon-277072/senec-workspace/documentation/4940eni/senec-app-api-4-3-3

And thats what i get:
`[
  {
    "id": "253699",
    "steuereinheitnummer": "S36700801093881874540162098",
    "gehaeusenummer": "DE-V3-H-10LI10-14556",
    "strasse": "Am xy",
    "hausnummer": "xyB",
    "postleitzahl": "xy",
    "ort": "xy",
    "laendercode": "xy",
    "zeitzone": "Europe/Berlin",
    "wallboxIds": [],
    "systemType": "V3"
  }
]`

Seems correct, if i am correct

Will send you an Mail with full Log

marq24, 2025-03-09:
and can you also do a request to:

https://app-gateway.prod.senec.dev/v1/senec/systems/{{SENEC_ANLAGE}}/technical-data

-> so in your case:
https://app-gateway.prod.senec.dev/v1/senec/systems/253699/technical-data

indikat0r, 2025-03-09:
`{
  "totals": null,
  "measurements": [
    "POWER_GENERATION",
    "POWER_CONSUMPTION",
    "GRID_IMPORT",
    "GRID_EXPORT",
    "BATTERY_IMPORT",
    "BATTERY_EXPORT",
    "BATTERY_LEVEL_IN_PERCENT",
    "AUTARKY_IN_PERCENT",
    "WALLBOX_CONSUMPTION"
  ],
  "timeseries": [
    {
      "date": 1640991600,
      "measurements": {
        "durationInSeconds": 31536000,
        "values": [
          0.16920936107635498,
          1086.5364990234375,
          1086.60546875,
          0.02621643990278244,
          2.7367539405822754,
          2.9678022861480713,
          0.16945771872997284,
          0,
          0
        ]
      }
    },
    {
      "date": 1672527600,
      "measurements": {
        "durationInSeconds": 31536000,
        "values": [
          13600.8232421875,
          16765.2265625,
          8288.7939453125,
          5007.1748046875,
          2322.04052734375,
          2213.539794921875,
          26.13227653503418,
          50.56,
          0
        ]
      }
    },
    {
      "date": 1704063600,
      "measurements": {
        "durationInSeconds": 31622400,
        "values": [
          12548.955078125,
          12657.4453125,
          7158.7001953125,
          9424.83984375,
          2871.38134765625,
          2746.64794921875,
          25.17909812927246,
          43.44,
          0
        ]
      }
    },
    {
      "date": 1735686000,
      "measurements": {
        "durationInSeconds": 8495930,
        "values": [
          1466.63671875,
          3709.685546875,
          2650.103515625,
          479.2021484375,
          969.95263671875,
          930.79150390625,
          32.14679718017578,
          28.56,
          0
        ]
      }
    }
  ]
}`

marq24, 2025-03-09:
my current assumption is, that "another" integration is messing up the shared ha websession - e.g. brute force deleting all cookies or other 'shit'... that an integrations should (IMHO) not do - or create/use an own/separate Websession...  

The log you shared with me is full of other error messages... I just can recommend, that you disable all current integration (or at least the ones you installed in the past few weeks) - and then check/confirm, that the webapi-integration return to normal operation.

indikat0r, 2025-03-09:
Thank you for looking over it.
I Think i will try it with a new install of Home Assistant but not now =)

closed 2025-03-25 by marq24