marq24/ha-senec-v3 #129: zwei Senec Instanzen in HA nach Änderung IP adresse

Zeppelin-PV, 2025-03-25:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

ich habe in HomeAssistant unseren Senec eingebunden und alles funktioniert
Vor kurzem habe ich die IP Adresse des Senec geändert und dies in HomeAssistant eingetragen
Nun habe ich zwei Instanzen stehen
die geänderte, die funktioniert
und ein die ist leer (keine Entitäten) und kann nicht entfernt werden
wie bekomme ich diese aus HA 

### Add your DEBUG log output

```shell

```

marq24, 2025-03-25:
möglicherweise musst Du beide Geräte in der GUI entfernen... (also nach einander) - aber ich muss gestehen ich habe k.A. ob das wirklich funktioniert... Vermutlich hat HA die gleiche DeviceID vergeben...

Zeppelin-PV, 2025-03-25:
das hört sich nicht gut an, zumal ja alles funktioniert,
die leere Instanz kann ich deaktivieren, aber nicht einzeln löschen
bei löschen werden vermutlich beide gelöscht
gibt es über den Editor eine Möglichkeit ?
ansonsten lasse ich das so,
DANKE, für die schnelle Antwort
[image]

marq24, 2025-03-25:
Ein Editor wäre mir nicht bekannt - man kann das 'sicherlich' im .storage in der devices.yaml patchen - das kann ich Dir aber auf keinen Fall raten - das musst Du für Dich entscheiden - diese zwei Einträge tun nicht weh - und ich schau' mal, ob ich da was bei einem der nächsten releases ggf. "fixen"/patchen kann...

marq24, 2025-03-25:
Noch ein kleines Update von mir zu dem Thema - Das Problem mit dem zweiten Eintrag ist u.a. auch dadurch entstanden, dass Du den identischen 'Namen' (`senec`) verwendet hast - das ist sagen wir mal 'suboptimal' das das möglich ist - diese Lücke muss ich auf jeden Fall schließen...

marq24, 2025-04-27:
hopefully fixed in 2025.4.0

Zeppelin-PV, 2025-06-14:
hallo
darf ich Dich um einen Gefallen bitten ?
wie kann ich in HA einen Zähler setzen der mir die NPU-Fehler anzeigt ?

so geht es auf jeden Fall nicht

alias: SENEC NPU Fehler
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.senec_system_state
    attribute: NPU-FEHLER
conditions: []
actions:
  - action: counter.increment
    metadata: {}
    data: {}
    target:
      entity_id: counter.npu_fehler_zahler
mode: single