marq24/ha-senec-v3 #130: Prevent non logical values for accu level

tklaas, 2025-04-22:
### With which version of the integration you have an issue?

SENEC.Home V4/SENEC.Home V4 hybrid

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I have a card showing the maximum acculevel per day for the past 30 days. 
I want to create another card showing the minimum level per day. 

But the minimum value is always 0 or below 1%.

I think senec is sending wrong values for some api requests or this plugin is maybe not getting a response and setting the value to 0?

In any case, it would be great, if the values could only be imported after a logic check. 
E.g. The accu level could not be 0% when it was 80% a minute earlier. 

Here you can see there are a lot of 0 values saved throughout the day
[image]

### Add your DEBUG log output

```shell

```

marq24, 2025-04-22:
I will double check this - I agree with you that it should be possible to check if the provided value by the fu****** stupid Senec API is plausible (or not).

But for here and now this is not a 🪲 

We will see, if I am in the right mood to support Senec Products in the Future where its so obvious thats all a big pile of shit... Have I mentioned, that my Senec System is not working for more then 4 Month?

marq24, 2025-04-23:
I spend some time with this... the question (code wise) is, when consider a value as possible "false"...

Ok starting with:
1) when value (provided by the API) = 0
2) when the previous value is not older then 15min
3) when the previous value is larger then 4 ( aka =5)

The conditions are highly depends from your configured update-interval - e.g. when polling only every 20min the API, then this code will fail...

tklaas, 2025-04-23:
is it possible to delete a previous value?
Then it may be possible to compare to the previous one.

1) 80%
2) 0%
3) 75%

Then the 0% could be deleted.

But this could go wrong depending on the update interval as well.

Assuming, that anyone with home assistant integration is interested in data that is as up-to-date as possible, a 15min intervall would not be configured. 
A possible way could be: 
Instead setting a fixed value to 15min, you can use something like 

- 5x the update intervall but a maximum of 20min

---
I had a look into my database in `statistics_short_term` which value is actually saved.
[image]

For a timestamp where the statistics shows 0%, the saved value is `1e-05` as minimum value.
This happens for max values as well and for mean, too, if both min and max are set to `1e-05`
[image]

So maybe filtering events with this value could be used to prevent false data.
If min OR max is wrong, use teh correct value for min and max, if both is wrong, just ignore the event

marq24, 2025-04-23:
... and I already altered my initial reply (using now 2 x updateInterval +30sec)

tklaas, 2025-04-23:
I had a look into `states` table, there is sometimes a value "unknown"
[image]

and `1e-05` seems to be the "normal" 0 value from senec
[image]

but also sends this value as false value:
[image]

> ... and I already altered my initial reply (using now 2 x updateInterval +30sec)
I think this would be enough and should match the most false values

marq24, 2025-04-23:
at least this seams to be a new phenomenon - somehow introduced at 13.04.2025 by one one more Senec Software Engineer Genius

[image]

as you see I have also this drops... but only 'once a day' ...

tklaas, 2025-04-24:
I don't think this is realted to the short term and long term statistics.
You can see the color of the line changing to a lighter blue indicating that data is coming from long term statictics. The single 0 values will not appear there when all other data is much higher.

As you can seen, in my chart it is beginning on 14.04.2025
I think if you have a look at it today, the drop on the 13.04. will not be visible anymore
[image]

tklaas, 2025-04-24:
You can simply see this by adding a statistics graph in a dashboard. When showing values for "minimum", the data should come from the raw data.
In my case everything was "OK" a year ago, then my Senec Home 4 was broken (so 0 values from then are correct)
With the new Home P4 installed at the beginning of december 2024, there are still 0 values every day

[image]

```
chart_type: bar
period: day
type: statistics-graph
entities:
  - sensor.senec_webapi_acculevel_now
stat_types:
  - min
hide_legend: true
title: "Akku minimum letzten 30 Tage "
logarithmic_scale: false
days_to_show: 500
```

marq24, 2025-04-27:
Please update to 2025.4.0... and close this issue when it's 'patched'

tklaas, 2025-04-27:
thanks for the quick feedback and implementation 👍 
I have updated the plugin and will have a look at the values and logs in a few days

marq24, 2025-04-28:
looks like the patch did NOT work (at least for me) - but I already have an idea why... 

edit the file `custom_components/senec/sensor.py`

search for line: 

`value == 0`

and replace it with

`value < 1 and value > 0`

save the file and restart HA

tklaas, 2025-04-30:
Everything looks fine for 2 days and i have a working chart 👍 

[image]

tklaas, 2025-07-02:
Seems like this is not working anymore with the newest version 2025.6.10

[image]

marq24, 2025-07-03:
@tklaas - can you please double check, if you find (again) debug log entries starting with:

`Thanks for nothing Senec! - API provided...`

in your HA debug logs ?!

marq24, 2025-07-03:
and it would be also great, if you can double check, that this "new" 0 (zero) values are again '1e-05' (and not 0) - might be this has changed as well ?!

marq24, 2025-07-03:
just used my short break for a refactoring session -> https://github.com/marq24/ha-senec-v3/releases/tag/2025.7.0

tklaas, 2025-07-04:
looks good, no zero values since yesterday ✅

[image]

tklaas, 2025-08-04:
yes, seems now senec is delivering 0 instead 1e-05

IN this previous comment you mentioned, that you planned to implement a logic to thrown non logic values away. Or was the first fix to just ignore non number values?
https://github.com/marq24/ha-senec-v3/issues/130#issuecomment-2823312463

marq24, 2025-08-04:
the problem is, "when" a jump to the value is acceptable? - this highly depends from your update-interval and the consumption...

tklaas, 2025-08-06:
again, thanks for fixing 🙏

marq24, 2025-08-06:
it's not really "fixing" - it's more - "einen Bogen drum rum machen"... But I am happy, when you are happy :-)