marq24/ha-senec-v3 #132: Cant login

alanodoug, 2025-05-15:
### With which version of the integration you have an issue?

SENEC.Home V4/SENEC.Home V4 hybrid

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hey
i have the new SENEC.Home P4 hybrid / 11.8
When I want to log in to the integration, i get an error message that my username or password is incorrect. 

### Add your DEBUG log output

```shell

```

alanodoug, 2025-05-15:
Hi,
yeah many times. Yes, I have tested the login several times and at the same time via the page. 
Since I got the change from V4 to P4, I had the problems that nothing was displayed to me anymore. So I deleted the Integration and wanted to do it again. And now I can’t get any further.

FrY-De, 2025-05-15:
I have the same issue, after Senec send a notification:

_"App Udate 4.8.0
Wir überarbeiten unseren Login, damit deine persönlichen Daten noch besser geschützt sind.
Mit dem Update auf die nächste App-Version 4.8.0 wirst du deshalb einmalig ausgeloggt, damit unsere Maßnahmen für dich in Kraft trerten."_

**translated:**
_"App update 4.8.0
We are revising our login so that your personal data is even better protected.
With the update to the next app version 4.8.0, you will therefore be logged out once so that our measures come into effect for you."_

marq24, 2025-05-15:
I just _can_ confirm that SENEC have changed their backend authentication - which makes the current implementation incompatible...

marq24, 2025-05-15:
please update to https://github.com/marq24/ha-senec-v3/releases/tag/2025.5.0 and close this if fixed - TIA

FrY-De, 2025-05-15:
Works perfectly for me.
Thank you very much.

closed 2025-05-16 by marq24