marq24/ha-senec-v3 #134: Web API stops reporting until reload since App 4.8.0

mschmiedel, 2025-05-21:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I have had it twice since the new App Version that at some point, the WebAPI stops refreshing data in HA. It feels like the login is lost and needs to be redone. Is anyone else experiencing this in the last few days? 

If so, maybe a relogin mechanism might be missing in case of Auth errors...

### Add your DEBUG log output

```shell

```

marq24, 2025-05-22:
since it's Senec - everything is possible - but currently the Webintegration is using the API-Endpoints of the 4.7 (old) App - so this "should" still work like normal...

Since we have here locally Internet & phone line issues I can't really say, IF it's the Senec Backend - or just some general network issues... I  just check and indeed my WebAPI Senec data also stopped updating somewhat yesterday...

mschmiedel, 2025-05-22:
Yep, I enabled debug in HA after the second time and I will be watching it. With 4.8, the login for all platforms was unified, and it feels like the tokens expire after 1 day or so (I also need to relogin in the Senec app every day). I am assuming this might be related, but let´s watch it...

marq24, 2025-05-22:
... cool - a SSO implementation for an App where the token expires in 24h - and an none working refresh - that's for sure what customers want's... 

but well - Senec customers are anyhow used to pain...

marq24, 2025-05-22:
... and just as addon - also the 4.8 app report here, that (after a relogin of course) my Senec V3 does not seams to be connected to the internet - with other word's THEY have not handled the requests the device is sending to the backend - looks to me like another total fuck-up of the system...

marq24, 2025-06-11:
check the debug log's if there are some messages "Thanks for nothing..."

spirit0607, 2025-06-11:
I have the same error. Here is my Log:
```
Logger: homeassistant.config_entries
Quelle: config_entries.py:963
Erstmals aufgetreten: 09:07:01 (6 Vorkommnisse)
Zuletzt protokolliert: 09:07:01

Error unloading entry senec_WEBAPI for button
Error unloading entry senec_WEBAPI for number
Error unloading entry senec_WEBAPI for select
Error unloading entry senec_WEBAPI for sensor
Error unloading entry senec_WEBAPI for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 963, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/binary_sensor/__init__.py", line 144, in async_unload_entry
    return await hass.data[DATA_COMPONENT].async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 200, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
```

marq24, 2025-06-11:
... `unloading...` das sollte "egal" sein... also das ist zu einem Moment, an dem die Integration aus HA entfernt wird - z.B. beim Beenden/Runterfahren von HA...

"ja" - das it trotzdem ein Fehler/Bug... aber das sollte nix mit dem Problem zu tun haben...

jrollenmueller, 2025-06-11:
I am having the same issue since today. Stopped working with my Senec Home P4.
I cleaned up (deleted) Senec Intrgration and tried to set it up again.

Now I am getting an error when tying to log in: "Login fehlgeschlagen - bitte Benutzername und/oder Passwort prüfen"

bobthebuilder4711, 2025-06-11:
Me too:

```
Logger: custom_components.senec.pysenec_ha
Quelle: custom_components/senec/pysenec_ha/__init__.py:3062
Integration: SENEC.Home ([Dokumentation](https://github.com/marq24/ha-senec-v3/blob/master/README.md), [Probleme](https://github.com/marq24/ha-senec-v3/issues))
Erstmals aufgetreten: 12:20:53 (79 Vorkommnisse)
Zuletzt protokolliert: 14:27:10

    APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/60432/measurements?resolution=YEAR&from=1514764800&to=1752236454 causing: 426, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/60432/measurements?resolution=YEAR&from=1514764800&to=1752236454'
```

https://mein-senec.de is up to date though.

marq24, 2025-06-11:
... 'thinking' or 'feelings' in the context does not help - either you have proof - then provide details - or relax and wait till someone with knowledge will take over...

In the meantime you might like to come over to my home and doing some garden and pool-setup work... then I might have some free time investigating the issue - as reported here #139