marq24/ha-senec-v3 #135: WebAPI polls some life-time values from wrong Senec

WernerSlabon, 2025-05-22:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [x] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I have two Senec Systems and added both to Home Assistant.
Some WEBAPI entities show the same values (lifetime values) whereby others are correct (current values)
See the image below:

[image]

This is an except of the log file. It looks like the values for each SENEC are retrieved/calculated.
But then the (higher) previous value of the first SENEC is compared with the second (lower) value and discarded!

### Add your DEBUG log output

```shell
[36m2025-05-22 02:01:50.002 DEBUG (MainThread) [custom_components.senec.sensor] SENSOR async_setup_entry[0m
[36m2025-05-22 02:01:50.004 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key consumption_total: 31761.7626748047[0m
[36m2025-05-22 02:01:50.004 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key consumption_total - prev:31761.7626748047 new: 26322.85942578125[0m
[36m2025-05-22 02:01:50.004 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key powergenerated_total: 35499.6759338379[0m
[36m2025-05-22 02:01:50.004 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key powergenerated_total - prev:35499.6759338379 new: 27184.404296875[0m
[36m2025-05-22 02:01:50.005 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key accuimport_total: 5755.43688964844[0m
[36m2025-05-22 02:01:50.005 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key accuimport_total - prev:5755.43688964844 new: 5755.4368896484375[0m
[36m2025-05-22 02:01:50.005 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key accuexport_total: 6031.81118774414[0m
[36m2025-05-22 02:01:50.005 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key accuexport_total - prev:6031.81118774414 new: 6010.24853515625[0m
[36m2025-05-22 02:01:50.006 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key gridimport_total: 13500.2507324219[0m
[36m2025-05-22 02:01:50.006 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key gridimport_total - prev:13500.2507324219 new: 10557.96630859375[0m
[36m2025-05-22 02:01:50.006 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key gridexport_total: 16805.3962097168[0m
[36m2025-05-22 02:01:50.006 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key gridexport_total - prev:16805.3962097168 new: 11104.3583984375[0m
[36m2025-05-22 02:01:50.019 DEBUG (MainThread) [custom_components.senec.switch] SWITCH async_setup_entry[0m
[36m2025-05-22 02:01:50.021 DEBUG (MainThread) [custom_components.senec.binary_sensor] BINARY_SENSOR async_setup_entry[0m
[32m2025-05-22 02:01:50.021 INFO (MainThread) [custom_components.senec.binary_sensor] No binary_sensors for WebPortal...[0m
[36m2025-05-22 02:01:50.021 DEBUG (MainThread) [custom_components.senec.button] BUTTON async_setup_entry[0m
[36m2025-05-22 02:01:50.021 DEBUG (MainThread) [custom_components.senec.number] NUMBER async_setup_entry[0m
[36m2025-05-22 02:01:50.022 DEBUG (MainThread) [custom_components.senec.select] SELECT async_setup_entry[0m
[32m2025-05-22 02:01:50.022 INFO (MainThread) [custom_components.senec.select] No selects for WebPortal...[0m
[36m2025-05-22 02:01:50.022 DEBUG (MainThread) [custom_components.senec.sensor] SENSOR async_setup_entry[0m
[36m2025-05-22 02:01:50.024 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key consumption_total: 31762.0810341797[0m
[36m2025-05-22 02:01:50.024 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key powergenerated_total: 35499.6759338379[0m
[36m2025-05-22 02:01:50.024 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key powergenerated_total - prev:35499.6759338379 new: 35499.67593383789[0m
[36m2025-05-22 02:01:50.025 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key accuimport_total: 5697.05683898926[0m
[36m2025-05-22 02:01:50.025 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key accuexport_total: 6031.81118774414[0m
[36m2025-05-22 02:01:50.026 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key gridimport_total: 13500.2517089844[0m
[36m2025-05-22 02:01:50.026 DEBUG (MainThread) [custom_components.senec.sensor] Thanks for nothing Senec! prev>new for key gridimport_total - prev:13500.2517089844 new: 13500.251708984375[0m
[36m2025-05-22 02:01:50.026 DEBUG (MainThread) [custom_components.senec.sensor] restored prev value for key gridexport_total: 16805.4059753418[0m
[36m2025-05-22 02:01:50.035 DEBUG (MainThread) [custom_components.senec.switch] SWITCH async_setup_entry[0m
```

marq24, 2025-05-22:
> Thanks for nothing Senec!...

IMHO this is not related to multiple Senec installations - it's a flaw in the backend of Senec discovered & reported a while ago (please search the issues)... And you can see in the log, that the two values are really nearby...

To my best knowledge in a Senec Systems Cluster one device is defined as __main__ (they call it still _master_) device - and this device communicate with all other systems and collect & summarize this data. 

Are you able to see the request/responses in the log too? [I am not 100% sure if this is enabled for my senec integration) - there you should be able to see, that there are different requests for your two instances of the integration.

But in general I am unsure, IF it's anyhow supported by Senec backend to query data from senec cluster separater,,, Since I do believe they are handled as a single Systemnummer...

marq24, 2025-05-22:
just a little update on this subject - since for what ever reason I have currently two MAIN Senec unit's in my MySenec Portal... 

When setting up the integrations for two different (independent) systems, the __key__ requirement is, that you:

- __must__ specify the 'Anlagennummer' manually (starting with 0 for the first, and 1 for the second)
- __must__ specify a different __name__ (you did that correctly in your example) - senec_4_WEBAPI & senec_5_WEBAPI

when this is done correctly, then you should see in the debug log lines like this:

```
 [custom_components.senec.pysenec_ha] APP-API get https://app-gateway.prod.senec.dev/v2/senec/systems/[YOUR_ANLAGEN_ID_HERE]/measurements?resolution=YEAR&from=1514764800&to=1750590625
```

when you setup everything correctly, then the request should appear 2 times (for both of the integrations - but with different [YOUR_ANLAGEN_ID_HERE] values... 

If you have in your logs only requests with the _identical_ ANLADGEN_ID - something in your integration setup is not correct.

WernerSlabon, 2025-05-22:
Actually the API-Calls and results are good.
I've deleted the "Senec_4_WebApi" device and recreated it with the name "Senec-4-WebApi".
This is the log output with the first data aquisition:

2025-05-22 16:46:09.420 INFO (MainThread) [custom_components.senec] Starting SENEC.Home Integration 'senec-4-WebApi' with interval:0:05:00 - ConfigEntry: {'created_at': '2025-05-22T14:46:09.419844+00:00', 'data': {'name': 'senec-4-WebApi', 'host': '<MASKED>', 'username': '<MASKED>', 'password': '<MASKED>', 'scan_interval': 300, 'type': 'web', 'dtype_int': '', 'dtype': 'SENEC WebAPI', 'dname': 'SENEC.HOME V3 hybrid duo LFP | SENEC.Num: DE-V3-HD-04LI10-92574', 'dserial': '<MASKED>', 'version': None, 'master_plant_number': 1, 'app_token': '<MASKED>', 'app_master_plant_id': '<MASKED>', 'app_wallbox_num_max': 1}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'senec', 'entry_id': '01JVW7AX8BXGN8BKMKRHK0CJFK', 'minor_version': 1, 'modified_at': '2025-05-22T14:46:09.419852+00:00', 'options': {}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'senec-4-WebApi', 'unique_id': None, 'version': 1}
2025-05-22 16:46:09.420 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: 'werner.slabon@slabon.de' with options: {'query_wallbox_data': False, 'query_spare_capacity': False, 'query_peak_shaving': False, 'app_wallbox_num_max': 1, 'app_token': '<MASKED>', 'app_master_plant_id': '<MASKED>'}
2025-05-22 16:46:09.420 DEBUG (MainThread) [custom_components.senec.pysenec_ha] MySenecWebPortal initialized and IntBridge.app_api set to <custom_components.senec.pysenec_ha.MySenecWebPortal object at 0x7fa58cbfe0d0>
2025-05-22 16:46:09.420 DEBUG (MainThread) [custom_components.senec.pysenec_ha] IntBridge.avail() -> IntBridge.app_api: True IntBridge.lala_cgi: True
2025-05-22 16:46:09.421 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** update_context(self) ********
2025-05-22 16:46:09.421 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********
2025-05-22 16:46:10.247 DEBUG (MainThread) [custom_components.senec] _async_update_data called
2025-05-22 16:46:10.264 DEBUG (MainThread) [custom_components.senec.pysenec_ha] got cookie update: senec-homev3-gui=c083f7c90db4b878d16f8e316151a9c704b1111e80333c989bc4827f63422e13;Secure;HttpOnly;Max-Age=3600;SameSite=Strict
2025-05-22 16:46:10.265 DEBUG (MainThread) [custom_components.senec] read: {'data': {'TEMPMEASURE': {'BATTERY_TEMP': 29.899999618530273, 'CASE_TEMP': 32.272552490234375, 'MCU_TEMP': 41.938323974609375}, 'PV1': {'POWER_RATIO': 100.0, 'POWER_RATIO_L1': 100.0, 'POWER_RATIO_L2': 100.0, 'POWER_RATIO_L3': 100.0, 'MPP_VOL': [418.58502197265625, 247.94500732421875, 425.8290100097656], 'MPP_CUR': [1.968000054359436, 0.0, 1.9490001201629639], 'MPP_POWER': [820.4100341796875, 0.0, 823.4710693359375]}, 'PWR_UNIT': {'POWER_L1': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L2': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'POWER_L3': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]}, 'PM1OBJ1': {'FREQ': 50.02000045776367, 'U_AC': [237.8000030517578, 237.0, 236.90000915527344], 'I_AC': [2.8299999237060547, 2.859999895095825, 0.26999998092651367], 'P_AC': [-642.1799926757812, -665.719970703125, 42.790000915527344], 'P_TOTAL': -1265.1199951171875}, 'PM1OBJ2': {'FREQ': 0.0, 'U_AC': [0.0, 0.0, 0.0], 'I_AC': [0.0, 0.0, 0.0], 'P_AC': [0.0, 0.0, 0.0], 'P_TOTAL': 0.0}, 'ENERGY': {'CAPTESTMODULE': [0.0, 0.0, 0.0, 0.0], 'GRID_POWER_OFFSET': 'VARIABLE_NOT_FOUND', 'GUI_BAT_DATA_COLLECTED': 1, 'GUI_BAT_DATA_CURRENT': -0.30000001192092896, 'GUI_BAT_DATA_FUEL_CHARGE': 100.0, 'GUI_BAT_DATA_MAX_CELL_VOLTAGE': 3469, 'GUI_BAT_DATA_MIN_CELL_VOLTAGE': 3451, 'GUI_BAT_DATA_POWER': -16.6200008392334, 'GUI_BAT_DATA_VOLTAGE': 55.400001525878906, 'GUI_BOOSTING_INFO': 0, 'GUI_CAP_TEST_DIS_COUNT': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_START': 'VARIABLE_NOT_FOUND', 'GUI_CAP_TEST_STATE': 0, 'GUI_CAP_TEST_STOP': 'VARIABLE_NOT_FOUND', 'GUI_CHARGING_INFO': 0, 'GUI_FACTORY_TEST_FAN': 'VARIABLE_NOT_FOUND', 'GUI_GRID_POW': -1265.1199951171875, 'GUI_HOUSE_POW': 395.381103515625, 'GUI_INIT_CHARGE_START': 0, 'GUI_INIT_CHARGE_STOP': 0, 'GUI_INVERTER_POWER': 1643.881103515625, 'GUI_TEST_CHARGE_STAT': 0, 'GUI_TEST_DISCHARGE_STAT': 0, 'INIT_CHARGE_ACK': 0, 'INIT_CHARGE_DIFF_VOLTAGE': 0.0, 'INIT_CHARGE_MAX_CURRENT': 0.0, 'INIT_CHARGE_MAX_VOLTAGE': 0.0, 'INIT_CHARGE_MIN_VOLTAGE': 0.0, 'INIT_CHARGE_RERUN': 0, 'INIT_CHARGE_RUNNING': 0, 'INIT_CHARGE_STATE': 0, 'INIT_CHARGE_TIMER': 0, 'INIT_DISCHARGE_MAX_CURRENT': 0.0, 'LI_STORAGE_MODE_RUNNING': 0, 'LI_STORAGE_MODE_START': 0, 'LI_STORAGE_MODE_STOP': 0, 'SAFE_CHARGE_FORCE': 0, 'SAFE_CHARGE_PROHIBIT': 0, 'SAFE_CHARGE_RUNNING': 0, 'STAT_HOURS_OF_OPERATION': 30989, 'STAT_LIMITED_NET_SKEW': 0, 'STAT_STATE': 13, 'TEST_CHARGE_ENABLE': 'VARIABLE_NOT_FOUND', 'TEST_CYCLE': 'VARIABLE_NOT_FOUND', 'ZERO_EXPORT': 0}, 'LOG': {'USER_LEVEL': 'VARIABLE_NOT_FOUND', 'LOG_IN_NOK_COUNT': 0}, 'BAT1': {'SPARE_CAPACITY': 0}, 'STATISTIC': {}, 'BMS': {'CELL_TEMPERATURES_MODULE_A': [27, 28, 29], 'CELL_TEMPERATURES_MODULE_B': [25, 27, 28], 'CELL_TEMPERATURES_MODULE_C': [0, 0, 0], 'CELL_TEMPERATURES_MODULE_D': [0, 0, 0], 'CELL_VOLTAGES_MODULE_A': [3462, 3463, 3459, 3454, 3457, 3459, 3461, 3461, 3462, 3455, 3458, 3458, 3463, 3454, 3454, 3458], 'CELL_VOLTAGES_MODULE_B': [3460, 3452, 3469, 3466, 3457, 3465, 3456, 3463, 3466, 3462, 3455, 3451, 3456, 3453, 3452, 3451], 'CELL_VOLTAGES_MODULE_C': [0, 0, 0, 0, 0, 0, 0, 0 [... cut ...]
Yesterday, I added the Senec-4 with Device-Index #0. After I saw that this is the Senec-5, I deleted it and created Senec-5 with Device-ID #0 and Senec-4 with Index #1.
I think at this point the Senec-5 values were stored "somewhere" (not using the entity name "Senec-4-WebApi" but something else that doesn't change).
Because total values shouldn't decrease, smaller values are discarded.
BUT where is the value from yesterday coming from and how to get rid of it???

Note: Todays value of Senec-5 is 31772 kWh, so the value "31761 kWh" is stored somewhere and doesn't get deleted, if I delete the whole entity.

WernerSlabon, 2025-05-22:
This was the trick!

Btw. Under Developer-Tools I just found an "entity value editor" with the entities.
After resolving the issue, I didn't see the "Senec_4_WebAPI" values.
Then I recreated the "wrong" entity and the old/false values were back.
Back to the entity editor, still the entities wern't listed.
Then after a reboot they were in the list and showing the false value.

Now, I set the value to 0 and waited until HomeAssistant refreshed the value.
The old/false value is back again!

Second try: I set the value to 0, REBOOT and waited until HomeAssistant refreshed the value.
Now, the correct value is shown (26359 kWh)

So, I assume that the last value is restored on reboot only and "cached" in memory ("sensor.py" class field "_previous_plausible_value".
Then this value is always returned (Line 163-170) because the new value is smaller.
So, to fix the values it is/was required to reboot before the next refresh was performed.

Huh... 
Tricky, tricky ...

closed 2025-05-22 by WernerSlabon