marq24/ha-senec-v3 #138: "sensor.senec_webapi_powergenerated_total" does not update anymore

fugazzy, 2025-06-11:
### With which version of the integration you have an issue?

WEB.API: mein-senec.de Portal

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

since yesterday 16:40 I do not get any actual data for this sensor anymore (its just constant at about 13.953,397 kWh)

### Add your DEBUG log output

```shell

```

marq24, 2025-06-11:
do you read the message/log output?!

__Thanks for nothing Senec__! prev>new for key gridexport_total - prev:8661.2353515625 new: 7317.4794921875

the last time the senec backend provided '8661.23' - now senec backend say the value is '7317.47' - this is INVALID for a total value...

closed 2025-06-11 by marq24

fugazzy, 2025-06-11:
OK, thank you, but how can I resolve that? Is that maybe because of the change of the akku modules (new LFP) last week?