marq24/ha-senec-v3 #139: YES - Senec.WEB.API has stopped working today (11.06.2025) @ 12 o'clock GMT!

marq24, 2025-06-11:
Keep Calm!

[image]

what ever Senec did, the chances are very likly, that either Senec realize, that some of the backen-systems is broken - or there might be an update of the __reverse engineered__ API-Access required...

No need to run around an panic - you are using a free service...

franzcasadei, 2025-06-11:
Maybe they switched to websockets (error 426)?

marq24, 2025-06-11:
_maybe_ you just be patient

marq24, 2025-06-11:
NIGHT-SHIFT is over - you really owe me one, two... ganz ganz viele...

ETH1166, 2025-06-11:
@marq24 Thank you very much for the immediate fix Matthias!!!
Works like a charm!

vb-raven, 2025-06-12:
I can also confirm thats working.

Thank you to @marq24

tklaas, 2025-06-12:
> NIGHT-SHIFT is over - you really owe me one, two... ganz ganz viele...
> 
> update to https://github.com/marq24/ha-senec-v3/releases/tag/2025.6.1

Thanks: anyone should read this and say thanks 🙏 😃 
https://github.com/sponsors/marq24

bobthebuilder4711, 2025-06-12:
Thank you for the quick fix.

All of my total values are much lower now though.

For example accuimport_total was 13.285kWh before. And mein-senec.de is currently at 13.293kWh, so that's plausible.

Meanwhile the integration with the fix now reports 45.350kWh...

marq24, 2025-06-12:
I don't want to sound un-polite - but if you don't understand the 'fix' of last night, you probably don't understand 'anything' else... 

IF you data is not plausible - it's SENEC's backend that's delivering this data - aka _Shit-IN -> Shit-OUT_ - the was __no change__ in the __requests or in the response-data processing__ of the integration! - it's now just communicating with a different Agent-Name...

vb-raven, 2025-06-12:
@electricdelicate the problem was/is communication with the senec server itself, not with the content in that communication.

vb-raven, 2025-06-12:
@bobthebuilder4711 what do you mean by numbers?

bobthebuilder4711, 2025-06-12:
> [@bobthebuilder4711](https://github.com/bobthebuilder4711) what do you mean by numbers?

[image]

See that massive jump down on all of those four numbers?

vb-raven, 2025-06-12:
These are values.

[image]

how I said......these are the same values as in the app

vb-raven, 2025-06-12:
[image]

Energy dashboard

PV in HA = 25.1
PV in App = 28.8

This is normal and of the end of a day are the same. This is korrelating with the Integralsensors

marq24, 2025-06-12:
As SENEC customer I have gave up the hope to have a fully functional energy-dashboard in HA with valid data... Way to many incidents with my SENEC System in the past... and I it's IMHO waste of time to try to fix/patch it afterwards.

hawk2004, 2025-06-12:
can anybody explain me how I add the new update in HA?

marq24, 2025-06-12:
> can anybody explain me how I add the new update in HA?

Please follow this routine:

1. In Home Assistant go to HACS
2. In the list of installed integrations search for `senec` 
3. Click on the 3-dot menu on the right side of the `senec` integration list entry
4. Click on Update Information
5. Note now that a new version is available
6. Install the update

marq24, 2025-06-12:
> [@marq24](https://github.com/marq24) Didn't the web part of the integration get its data from mein-senec.de initially? Can I ask why you changed it to the App Api?

... I can't remember to be honst - IMHO the main reason was, that it stopped working...

marq24, 2025-06-12:
> Senec change the login mechanism

I am sorry - but this is __not__ correct... look into the changes of the commit/fix - THAT was enough to 'solve' the problem... No alternative login, no new SSO, or other data sources then before... https://github.com/marq24/ha-senec-v3/commit/f76b988ba4dbb9245b491bea1ebb5442a1f8575d

bobthebuilder4711, 2025-06-12:
@marq24 I'm asking because I can still inspect the site and see that it polls getstatus,php which returns JSON that includes the correct total value. I might have to look into getting that back into the integration because I don't like that the App Api is changing constantly. Biggest roadblock for me is how HA works behind the scenes.

@vb-raven I feel like you're guessing here. mein-senec.de was the initial source for the web part of this integration here. I know because I wrote the proof of concept together with mstuettgen. And I'm only not using that anymore because my home assistant instance crashed the hard drive and when setting up a new instance it was faster to simply use marq24's integration. But the addon on mstuettgen's github should still work since you still need to poll getstatus.php?type=accuimport&period=all&anlageNummer=0 etc for the total values.

vb-raven, 2025-06-12:
@bobthebuilder4711 for me it works :-)....... I have connection (at the moment) and the **values** are right (app and HA the same) and I believe the korrelation in der Energy Dashboard at the end of the day are also right.
You need a M2 SSD.....a micro SD Card or normal usb stick I will give half a year with home assistant.

And you saw the log I have pasted. The battery is communicating over an api. the frontend can be a website or a third-party plugin like this here or clients like postman or insomnia.

bobthebuilder4711, 2025-06-12:
@vb-raven I've got a whole computer with a dedicated SSD for my HA instance. Thank you, but I know what I'm doing.

> And you saw the log I have pasted. The battery is communicating over an api. the frontend can be a website or a third-party plugin like this here or clients like postman or insomnia.

You're mixing stuff up here. It's not the battery but the home assistant integration that communicates with the senec servers via the App API. The frontend for the App Api is the official App. The mein-senec.de website isn't using the App API but its own API. And postman isn't a client but a site where you can publish APIs.

No clue what "insomnia" is.

Thomk-63, 2025-06-12:
@marq24 Thank you for the quick fix of the issue with SenecWebApi! Now every energy counter works again, with 1 exception: sensor.senec_webapi_consumption_total. This is still fozen and is not increasing anymore.
Any idea, what the reason might be?

electricdelicate, 2025-06-12:
Whats happened here?
Numbers from now, as you see, until the broken access was everything fine after that, the values are wrong.

[image]

vb-raven, 2025-06-12:
@electricdelicate do you use the integration with Home Assistant or at your own?

marq24, 2025-06-12:
... I close this here - since I don't want to fuel the troll-like discussion... I have gave up the hope, that people search, read and listen to available information - instead of asking the same questions again, again & again... 

1. There was a global outage of the integration cause of a change at SENEC backend infrastructure (some sort of WAF) - where they decided to block current access - any other speculation does not lead to anywhere - the blocking-element has been adjusted last night - and since then 'everything' was return to normal operation - at least concerning the core functionality.

2. For what ever reason SENEC APIs does not deliver every time the same data - specially for the overall counters - here I only can speculate about the possible reasons: _might be_  that's just a simple failure in the backend database servers, _might be_ invalid single entry's, _might be_  cause of statistic updates & recalculations or _might be_ cause of simple backup & recovery procedures. So many things can go wrong, why data that the SENEC-API is going to provide might not be the values previously provided via the identical API calls - at the end of the day I do not have any kind of information about the __WHY__. It's just a __fact__ that this is happening from time to time... But this is no __news__ I am not sure if this started with the replacements of the battery modules for some users - or if this was also just co-incidence. 

3. There are two concurrent backends @ SENEC - one is the mein-senec.de portal where you might be able to access statistic data from your systems [to my best knowledge the mein-senec.de portal is not available to all SENEC Customers] - and then there is the SENEC Mobile App - for what ever reason the data of this two places __can differ__ from each other for __some__ users.  What's the root cause of such a possible gap is unknown. Of course a gap does not make any sense - but with some patience there are chances that this data will be harmonized (either by Hand or automated processes running @ SENEC) 

4. The global usage counters of this Integration are configured to have __ONLY__ one direction - and this ONE direction is only __UP__. There is no way that total kWh power consumption counter can decrease (as long as you are not able to do time travel / or manipulate your power meter at home) - Therefore a LONG time ago it has been requested (& implemented) that the total counters ONLY can go UP. Use the search function to get additional information about this here @ GitHub. Each time the Integration reads such a value (that is lower then a previous one) you will find log entries for this type of event - also this have been explained more the one time at different places - also how to enable debug logging is well documented

5. YES - inconsistent total counters going to mess up the HA Energy Dashboard - as said earlier, als long as you have no alternative data sources and you rely on SENEC backend data - switch to alternatives like the RiemanSum Integral - or accept the fact, that's just fucked up.

electricdelicate, 2025-06-12:
That came also in my mind and I've done this already. But  I've recognized an another issue, since this stupid action from senec the values don't change anymore although  my SENEC has definitely charged and discharged the last hours.

You too?

bobthebuilder4711, 2025-06-13:
Looking at the log, I've got multiple errors related to "_previous_plausible_value_ts". So I'm guessing that the values have jumped even lower and now they need to catch up before the integration lets them through again.

But there is no way of seeing the actual value received by the API, so finding the root cause isn't that easy.

marq24, 2025-06-13:
@borg16 thanks for your evaluations! - since my MCU was replaced, I have now "of course" two SENEC Systems in the backend... I will figure out over the weekend, if a simple sum-up could be implemented... [and of course calling all previous years only once... even if I must assume, that in SENECs universe even data from the past going to change... ]

Have that said - I would probably have the need of an Login/credentials of such a "old" SENEC System...

bobthebuilder4711, 2025-06-13:
Could it be that the two years are a moving window? Meaning if I ask for the current total value at noon then the oldest value will be from noon two year ago?

Because the increase per hour on home assistant and what I can see on mein-senec.de don't match up at all for any of the 6 "total" sensors.

So it's not just some absolute shift.

---

In the meantime I've deleted the web part of the integration, have replaced the missing sensors with template sensors and have them pointed at the mqtt sensors from the senecweb2mqtt addon. Because that still works with correct data. (But I'm using an updated version that has http request timeouts.)

---

@marq24 I'm not so keen on giving out my credentials but maybe there is some other way? Maybe a simple python script that includes what you want to test and that I could execute?

marq24, 2025-06-13:
> I'm not so keen on giving out my credentials but maybe there is some other way...

that's totally reasonable - IF, then the procedure would be anyhow:
Change your current password...  get in contact with my by eMail, let me know the login, I will provide you with an alternative channel (SMS) to submit finally the pwd (of you can use tools like Bitwarden/Lastpass to share credentials) - I'll do my work, once that is done, you change back again the pwd to the previous one...

The main issue is, that I currently does not know, for what exactly I am looking for in the data... But again - there might be other alternatives... I have here no pressure - since I am into another Game with Ford right now :-D - They are at least at Level 20-30 ;-)

bobthebuilder4711, 2025-06-13:
Let's rephrase that: I'm not so keen on giving someone else my account.

I'm just someone who's got a V2.1 that has been running since 2019, so more than two years.

But no clue how long that will still be true. The system has been throttled down to 50% since the beginning of April. So we're probably also getting the LFP packs somewhere in the future and I think I remember that people with a V2.1 get a replacement V3 with the LFP packs.

I'm also not in any rush. senecweb2mqtt is currently working.

spirit0607, 2025-06-14:
BTW: As an alternative to the Energy Dashboard, I can strongly recommend https://solectrus.de In this Dashboard, there are many more opportunities than in HA. For SENEC Users, there is also an importer for historical energy consumption.

Disincentivato, 2025-07-07:
Hi, yes, It seems like it happened again. No update from the webapi from yesterday. :)

closed 2025-07-15 by marq24