marq24/ha-senec-v3 #141: Spare Capacity

Home-KN-LE, 2025-06-15:
Hello, 
For some time now, the reserve capacity (spare capacity) value hasn't been able to be read and controlled. Could this 
function be restored/repaired? In conjunction with Tibber, 
this could prevent discharging at least from 50% to 0%, allowing only low-cost electricity to be used directly. 

Thanks. Mike

marq24, 2025-06-15:
@io-debug - would you have time to check, IF I have messed this up with the latest patches? [since I can't test this here at all with my system]

Home-KN-LE, 2025-06-15:
Hello,

Thanks for your reply. I've been postponing the update since version 2014.12.1, because starting with the next version (i.e., 2024.12.2), only data with "unknown" for the spare capacity was displayed. However, since June 11, 2025, nothing has been displayed at all, and I ran the update anyway. Now the other data is displayed again, but again with "unknown" for the spare capacity...
What data can I help you with?

Regards, Mike

marq24, 2025-06-15:
@io-debug - thanks for the input,
then `datetime.fromtimestamp(...)` is the issue - since we do not provide tz info in the call - I will make a quick fix with tz=UTC which is probably not correct in the case of the SENEC system

marq24, 2025-06-15:
@Home-KN-LE  please update to https://github.com/marq24/ha-senec-v3/releases/tag/2025.6.4 and if issue is solved, please close it @io-debug - at some day we must check the timzone information (I guess we now just display the end time in GMT - but actually SENEC backend wanted to provide the info in 'local' timezone...

io-debug, 2025-06-15:
@marq24 

_Timezone-Issue
I dont see any log entry regarding this issue anymore. Looks good to me.

_Spare Capacity
I've provided an PR.
From my point of view there was an wrong check. (Is the spare capacity an int/float).
The check took place, bevor the text was converted into an int.
My local test worked fine, therfore I would give it a try.

marq24, 2025-06-15:
sorry for closing this too early

Home-KN-LE, 2025-06-18:
Hallo marq24,
also erst einmal Hut-AB für deine Leistung im Zusammenhang mit diesem Adapter, du schaffst sicher eine Lösung, alles Gut!,
die Version 2025.6.8 habe ich installiert, aber leider, wird nun ständig 0% angezeigt, eine Änderung im SenecPortal wird nicht übernommen, ein deaktivieren/aktivieren oder Neustart von HA bringt auch keine Lösung, schaust du bitte noch einmal.
Vielen Dank für deine Bemühungen.
Schönen Abend!
;)

io-debug, 2025-06-18:
Hi, 
bei mir war es ebenfalls der Fall, dass mit dem letzten Update die Reservekapazität (Spare Capacity) mit "0" angezeigt wurde. ich habe ein Pull Request erstellt, welches wieder dafür sorgen sollte, dass das Spare Capacity angezeigt wird. 

Zu dem Thema, dass Anpassungen aus dem Senec Portal nicht übernommen werden:
Als wir die Steuerung der Reserverkapazität implementiert haben, haben wir das Feature so implementiert, dass wir nur alle 24 Stunden die Daten bei Senec abrufen.
Grund/Ziel: Wir halten die Anzahl an Anfragen gering und werden nicht geblockt, weil wir zu viele Anfragen stellen.

Für deinen Fall ist das natürlich ungünstig - Hier müssen wir einmal überlegen, was ein sinnvoller Weg sein könnte...

marq24, 2025-06-18:
@io-debug Danke für Deinen erneuten PR in der Sache... wie kann das aber keine "Number" sein was da als Response zurückkommt... ist das wirklich ein "plain-string" in dem dann eine Zahl steht... also nix JSON like... traurig... 

Ich habe das jetzt nochmal modifiziert und nochmal extra ein handling für ein ValueError gemacht, weil da kann ja auch was ganz anderes in der Response stehen - also jeder beliebige Text... und damit dann der Code nicht crashed habe ich eben nochmal die Modifikation gemacht... wäre cool, wenn Du das nochmal ausprobieren kannst, bevor ich wieder ein weiteres 'fail' release mache...

io-debug, 2025-06-19:
Kurz und Knapp aus der Mittagspause:

- @marq24 danke für deine Ergänzung um das Error Handling - da hast du natürlich Recht
- Nach meinem wissen müsste "response.text()" einen String zurückgeben
- Ich habe deine Anpassungen fix getestet - funktioniert bei mir

Danke dir!

PR #142 (2025-06-15): Update __init__.py
Should close #141 

The Change removes the check "isinstance(self._spare_capacity, (int, float)", because the text is converted after the check into an int. Therefore the check must fail and the spare capacity is not shown to the user.
commit 0b90d400 (2025-06-15): Update __init__.py

Should close #141.

The Change removes the check "isinstance(self._spare_capacity, (int, float)", because the text is converted after the check into an int. Therefore the check must fail and the spare capacity is not shown to the user.
closed 2025-06-21 by Home-KN-LE