marq24/ha-senec-v3 #149: "Akku Volladung" do not work anymore

electricdelicate, 2025-07-11:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I cannot call the "Akku Volladung" anymore at my Senec v2.1! Is that a new bug or a new restriction by Senec?

### Add your DEBUG log output

```shell
2025-07-11 21:52:13.364 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139674276145536] Error during service call to switch.turn_on: 
2025-07-11 21:52:16.496 WARNING (MainThread) [custom_components.senec] UpdateFailed (fatal): 'str' object has no attribute 'copy'
2025-07-11 21:52:16.496 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139674276145536] Error during service call to switch.turn_off: 
2025-07-11 21:52:18.201 DEBUG (MainThread) [custom_components.senec] _async_update_data called
```

marq24, 2025-07-11:
this looks like a bug I must have introduced in one of the latest releases - let me check

marq24, 2025-07-11:
please update to https://github.com/marq24/ha-senec-v3/releases/tag/2025.7.6 and close when fixed - thanks for your report!

electricdelicate, 2025-07-12:
Works like a charm!