marq24/ha-senec-v3 #150: Senec Web Api stopped reporting Energy consumption counters totay (14.7.25) at 9:32

Thomk-63, 2025-07-14:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Since today at 9:32 the Senec WebApi has stepped to report enegry consumption counters such as senec_webapi_consumption_total.

### Add your DEBUG log output

```shell
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.senec.pysenec_ha
Quelle: custom_components/senec/pysenec_ha/__init__.py:3179
Integration: SENEC.Home (Dokumentation, Probleme)
Erstmals aufgetreten: 17:44:48 (21 Vorkommnisse)
Zuletzt protokolliert: 18:01:23

APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551723 causing: 404, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551723'
APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551813 causing: 404, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551813'
APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551903 causing: 404, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551903'
APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551993 causing: 404, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752551993'
APP-API: Could not get data from https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752552083 causing: 404, message='', url='https://app-gateway.prod.senec.dev/v2/senec/systems/262615/measurements?resolution=DAY&from=1752451200&to=1752552083'
```

marq24, 2025-07-15:
IMHO The output of the SmartMeter is not able to provide you with the sufficient data - you need a PowerMeter measuring the complete consumption of your home - independent from a possible battery or PV feed... And then you need to be able to measure, what's happening between your home and the battery... I don't want to say this is "impossible"... But switching over to [RiemanSumIntegral ](https://www.home-assistant.io/integrations/integration/) Helper would be way simpler

lennartschulte, 2025-07-15:
My idea is to calculate the kwh based on the live data provided by the senec unit local. it still gives me the House consumption and battery consumption.

[image]

i would try to combine these values with the readings of the smart meter to ensure that i'm not missing data while homeassistant is offline etc.

best case screnario would be if senec changes their policy regarding the api output.

marq24, 2025-07-15:
want to use senec.local data? [not the option for Senec V4+ users - I know] -> https://github.com/marq24/ha-senec-v3/issues/4

eiderterra, 2025-07-15:
Hello,
I was very enthusiastic about the integration so far, but now my questions: will the data from senec be delivered again, will the performance data, e.g. "SENEC WebAPI: senec_WEBAPI Current battery discharge" still be available so that a calculation with the Riemann sum integral sensor makes sense at all, or would it be better to use your own measuring system, such as Shelly 3mPro.

eiderterra, 2025-07-15:
> Right now - the only option would be to query the mein-senec.de webportal (like in the initial WebAPI implementation) 

please can you explain this to me

marq24, 2025-07-15:
> Do you have any recommendations for the integral settings?

@ETH1166 do you have some hint's to share?

bobthebuilder4711, 2025-07-19:
To prevent having to switch sensors for the energy dashboard, you should think about using templates. That way you can always switch data sources without messing up your statistical/historical data.

For example this is my accu export sensor:
```
template:
  - sensor:
    - name: "senecweb_accuexport_total"
      unique_id: "senecweb_accuexport_total"
      unit_of_measurement: "kWh"
      device_class: "energy"
      state_class: "total"
      icon: "mdi:home-battery"
      state: >
        {{ states('sensor.senecweb2mqtt_accuexport_total') }}
```

I created those the last time the App Api didn't work. I deleted the "web" part of the integration, switched back to the senecweb2mqtt solution and then created those template sensors so the statistical/historical data is preserved.

And btw: polling mein-senec.de is still working as usual. This is from today since midnight: 

[image]

marq24, 2025-07-19:
> And btw: polling mein-senec.de is still working as usual. 

if you are interested you might like to take a closer look into the sources of the current release concerning polling (again) mein.senec.de...

bobthebuilder4711, 2025-07-19:
Looking at the sources it seems like you're still logging into mein-senec.de to poll other stuff but then you're not updating any total values.

Seems like the only thing that needs to be done is to uncomment line 3878 in ha-senec-v3/custom_components/senec/pysenec_ha/\_\_init\_\_.py:
`# await self.web_update_total()`

marq24, 2025-07-19:
> Edit: The sources I see in github and the code that gets installed, seem to differ from each other...

letme double check...

bobthebuilder4711, 2025-07-19:
I don't need this. I've got better things to do than to get my work of the past two hours deleted...

marq24, 2025-07-19:
> I don't need this. I've got better things to do than to get my work of the past two hours deleted...

I am really sorry - please don't get upset... I am anyhow struggling with all this

kopierschnitte, 2025-07-20:
I totally understand both of you and in my opinion, this is another proof of why to avoid "cloud-only" services. I'm so glad I don't have one of the newer Senec batteries.

@marq24 Would you mind telling, what you're concerned of?

marq24, 2025-07-20:
> [@marq24](https://github.com/marq24) Would you mind telling, what you're concerned of?

Du kannst mich gerne per eMail kontaktieren...

marq24, 2025-07-23:
> About the changing signature, is it some kind of nonce token? I will take a look at the senec api in the coming days, maybe I can help with the mechanics of the login

Thanks for your offer - help is ALWAYS appreciated - in the current SENEC case I do believe I have already everything that is needed -   and yes 'nonce' is part of it [but IMHO without any function] - but let's skip a discussion about this subject for now...

> At the end, if the browser is able to login, it must 100% be possible to do the login with a script... ;)

oh - well - I am not 100% confident about this statement - since month I try to login @ ford.de via a script... But all the JS magic is - well it is really magic for me... So when you have a script, that is able to open this login dialog - and successfully submit login data - it would be great! - https://www.ford.de/mein-ford-account/