marq24/ha-senec-v3 #159: Senec wallbox integration works only limited

julianrehm87, 2025-08-01:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

The wallbox charging modus is not working anymore. When changing it in either Home Assistant or Senec App, both ways dont update each other anymore. Also the current wallbox power (ladeleistung in W) is not working anymore. Neither the Senec.Home V3 hybric integration nor the WebAPI integration deliver a value anymore.

### Add your DEBUG log output

```shell
2025-08-01 15:31:55.791 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): requesting: https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search
2025-08-01 15:31:55.802 ERROR (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): Error while access https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search: '405, message='', url='https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search'' - Response is: '<ClientResponse(https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search) [405 None]>
<CIMultiDictProxy('Date': 'Fri, 01 Aug 2025 13:31:55 GMT', 'Content-Type': 'application/json', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Allow': 'POST', 'X-Content-Type-Options': 'nosniff', 'X-XSS-Protection': '0', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'Strict-Transport-Security': 'max-age=31536000; includeSubDomains', 'X-Frame-Options': 'DENY')>
'
2025-08-01 15:31:55.802 WARNING (MainThread) [custom_components.senec.pysenec_ha] app_update_all_wallboxes(): No valid wallbox data received or data is not a list None
2025-08-01 15:31:55.802 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** web_update_peak_shaving(self) ********
2025-08-01 15:31:55.835 ERROR (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): Error while access https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search: '405, message='', url='https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search'' - Response is: '<ClientResponse(https://senec-app-wallbox-proxy.prod.senec.dev/v1/systems/wallboxes/search) [405 None]>
<CIMultiDictProxy('Date': 'Fri, 01 Aug 2025 13:31:55 GMT', 'Content-Type': 'application/json', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Allow': 'POST', 'X-Content-Type-Options': 'nosniff', 'X-XSS-Protection': '0', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'Strict-Transport-Security': 'max-age=31536000; includeSubDomains', 'X-Frame-Options': 'DENY')>
'
2025-08-01 15:31:55.835 WARNING (MainThread) [custom_components.senec.pysenec_ha] app_update_all_wallboxes(): No valid wallbox data received or data is not a list None
2025-08-01 15:31:55.835 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** web_update_peak_shaving(self) ********
2025-08-01 15:31:55.929 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** app_update(self) ********
2025-08-01 15:31:55.929 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_system_details(self) ********
2025-08-01 15:31:55.929 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: _app_do_get_request(self) ********
2025-08-01 15:31:55.929 DEBUG (MainThread) [custom_components.senec.pysenec_ha] app_verify_token(): access_token is VALID [expires in: 2m 54s]
2025-08-01 15:31:55.929 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): requesting: https://senec-app-systems-proxy.prod.senec.dev/systems/193373/details
2025-08-01 15:31:55.958 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** app_update(self) ********
2025-08-01 15:31:55.958 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_system_details(self) ********
2025-08-01 15:31:55.958 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: _app_do_get_request(self) ********
2025-08-01 15:31:55.958 DEBUG (MainThread) [custom_components.senec.pysenec_ha] app_verify_token(): access_token is VALID [expires in: 2m 54s]
2025-08-01 15:31:55.958 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): requesting: https://senec-app-systems-proxy.prod.senec.dev/systems/193373/details
2025-08-01 15:31:56.093 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): response: {'systemOverview': {'systemId': 193373, 'hid': None, 'productName': 'SENEC.HOME V3 hybrid duo LFP', 'installationDateTime': '2021-09-29T11:31:22Z', 'exchangedSystemId': None}, 'casing': {'serial': 'DE-V3-HD-03LI10-55796', 'temperatureInCelsius': 30.899048}, 'mcu': {'mainControllerSerial': 'S47841771414942728540031043', 'mainControllerUnitState': {'name': 'NETZ_UND_ENTLADEN', 'severity': 'INFO', 'operatingMode': 'FULL_OPERATION'}, 'firmwareVersion': '833', 'guiVersion': 970, 'lastContact': {'timestamp': '2025-08-01T13:27:52.167Z', 'severity': 'INFO'}, 'ipAddress': None}, 'warranty': {'endDateTime': '2031-09-29T22:00:00Z', 'warrantyTermInMonths': 120}, 'batteryPack': {'numberOfBatteryModules': 2, 'technology': 'LITHIUM_IRON_PHOSPHATE', 'currentChargingLevelInPercent': 97, 'maxCapacityInKwh': 10.0}, 'batteryModules': [{'ordinal': 1, 'serialNumber': '0176334', 'state': {'state': 'OK', 'severity': 'OK'}, 'lastContact': {'timestamp': '2025-08-01T13:27:51Z', 'severity': 'OK'}, 'firmwareVersion': None}, {'ordinal': 2, 'serialNumber': '0186810', 'state': {'state': 'OK', 'severity': 'OK'}, 'lastContact': {'timestamp': '2025-08-01T13:27:51Z', 'severity': 'OK'}, 'firmwareVersion': None}], 'installer': {'companyName': 'Solarstromer Inh.: Manfred Engisch', 'email': 'info@solarstromer.de', 'phoneNumber': '+49 7444 917846', 'address': {'street': 'Schillerstraße', 'houseNumber': '49', 'postcode': '72275', 'city': 'Alpirsbach', 'countryCode': 'DE', 'region': 'Baden-Württemberg', 'timezone': 'Europe/Berlin'}, 'onlineMonitoringAllowed': True, 'website': 'http://www.solarstromer.de'}}
2025-08-01 15:31:56.093 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_dashboard(self) ********
2025-08-01 15:31:56.093 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: _app_do_get_request(self) ********
2025-08-01 15:31:56.093 DEBUG (MainThread) [custom_components.senec.pysenec_ha] app_verify_token(): access_token is VALID [expires in: 2m 53s]
2025-08-01 15:31:56.093 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): requesting: https://senec-app-measurements-proxy.prod.senec.dev/v1/systems/193373/dashboard
2025-08-01 15:31:56.122 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): response: {'systemOverview': {'systemId': 193373, 'hid': None, 'productName': 'SENEC.HOME V3 hybrid duo LFP', 'installationDateTime': '2021-09-29T11:31:22Z', 'exchangedSystemId': None}, 'casing': {'serial': 'DE-V3-HD-03LI10-55796', 'temperatureInCelsius': 30.899048}, 'mcu': {'mainControllerSerial': 'S47841771414942728540031043', 'mainControllerUnitState': {'name': 'NETZ_UND_ENTLADEN', 'severity': 'INFO', 'operatingMode': 'FULL_OPERATION'}, 'firmwareVersion': '833', 'guiVersion': 970, [... cut ...]
2025-08-01 15:31:58.059 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_dashboard(self) ********
2025-08-01 15:31:58.059 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: _app_do_get_request(self) ********
2025-08-01 15:31:58.060 DEBUG (MainThread) [custom_components.senec.pysenec_ha] app_verify_token(): access_token is VALID [expires in: 2m 51s]
2025-08-01 15:31:58.060 DEBUG (MainThread) [custom_components.senec.pysenec_ha] _app_do_get_request(): requesting: https://senec-app-measurements-proxy.prod.senec.dev/v1/systems/193373/dashboard
```

marq24, 2025-08-01:
Have you read any of the release notes???

marq24, 2025-08-02:
> 1. Why are certain parameters of the wallbox still updating, e.g. if a car is connected or not and others like power consumption not (cannot evaluate this technically).

Seams to be not only technical limitations - I have the impression there is _no understanding_ how much time & effort must be spend maintaining such a project, how testing could work, for people willing to support - specially in the wallbox case (since my setup is a different one)...

The created __bug__ issue really really piss me off - I can't phrase it different - it's stated with every release that wallbox is not supported yet! 

You don't have a single clue concerning the amount of time I have spend in the past days with this project - your don't understand technically what's going on - but you file your expectation: something should either __be fixed now__ or should be removed __10 days after__ an initial release - for a service/software you __haven't paid anything__.

What you can do _better_ next time? You could have started a discussion, offering your help, willing to test & report to support a return of a wallbox implementation in the future.

Ich geh jetzt Frühstücken

closed 2025-08-02 by marq24

julianrehm87, 2025-08-02:
> Seams to be not only technical limitations - I have the impression there is _no understanding_ how much time & effort must be spend maintaining such a project, how testing could work, for people [...]

Hi Marc,

I am really sorry if this pissed you off. This was really not my intention, really! I understand you spend a lot of time in this and as i said in my last message I also really appreciate this that there are people like you who do this. I also didnt want to say that this must be fixed. I just thought it would be consistent to hide it (like you also did with the totals sums in between).

I will take the discussion way next round, but honestly I am not a regular github user so I am not so familiar with all options.

Again, sorry, if this was misunderstood by you in the wrong way.

Julian