marq24/ha-senec-v3 #161: Falscher Gesamtwert bei der PV-Produktion

Schlolle, 2025-08-03:
Der Wert meiner Entität "sensor.senec_webapi_powergenerated_total" Solarproduktion Gesamt war bis zum 12. Juni 2023 korrekt. Dann machte dieser zunächst einen Sprung nach unten, war dann einen Tag wieder korrekt, machte dann einen Sprung nach oben, um vom 6. - 10. Juli nochmals korrekt dargestellt zu werden. Seither ist er permanent zu hoch. (siehe Diagramm). Aktuell liegt er bei 64038 kWh, korrekt wären 36.768 kWh. Habe nur ich das Problem? Wie könnte man es korrigieren? In den Stastiken habe ich es schon über einen einmaligen Negativwert versucht - das hat nicht funktioniert.
[image]
[image]

marq24, 2025-08-03:
Du hast aber schon den ganzen Trubel um die SenecAPI die 632 releases der Integration und den restlichen Affenzirkus mitbekommen - oder?

Du kannst mal im HA ./strorage Verzeichnis den neu angelegten Senec cache löschen (einfach die Datei im 'senec' Unterordner) - wenn das nicht funzt - 'müsste' ich einmal selbst hier lokal debuggen - dazu bräuchte es aber viel Vertrauen auf Deiner Seite - Du kannst mich gerne per eMail kontaktieren.

Schlolle, 2025-08-04:
> ... alternative zum manuellen löschen gibt's auch in der Integration einen neuen Button, den Du aber vorher erstmal aktivieren müsstest

Das mit dem neuen Button hat leider nicht funktioniert. Aktuell ist es bei mir jetzt so, dass heute der Wert gar nicht mehr erhöht wurde. Siehe Diagramm

Dadurch kann auch der Autarkiegrad momentan nicht mehr berechnet werden.

bobthebuilder4711, 2025-08-06:
Vielleicht wäre eine Einstellung gut, dass man die Abfrage aus machen kann, ob die Werte niedriger sind als vorher. Dann kann man zumindest in der Statistik den Sprung auf 0 setzen und wenn es wieder hoch springt, kann man auch diesen Sprung auf 0 setzen und hat dadurch brauchbare Werte in der Statistik.

(Klar, für die aktuelle Datenaufzeichnung hilft das nicht, aber die ist ja sowieso nur 2 Wochen da und wird dann gelöscht und für das Energy Dashboard wird nur die Statistik benutzt.)

Wenn der Nutzer aufgeklärt ist, dann ist eine Korrektur solcher Sprünge auch kein großes Problem mehr. Im "Statistik anpassen" Fenster gibt es sogar einen extra "Ausreißer" Knopf, damit man sowas schnell findet und korrigieren kann.

marq24, 2025-09-07:
Ich denke wir können den hier schließen?

Schlolle, 2025-09-07:
Ja, der Wert ist wieder korrekt! Vielen Dank.