marq24/ha-senec-v3 #171: Multiple SENEC system support broken

WernerSlabon, 2025-09-14:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [x] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [x] I confirm that I have [read & understand the main Integration 'check-list'](https://github.com/marq24/ha-senec-v3/discussions/165)
- [x] I have installed the **latest** release version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainer of this repo will spend time explaining the world to me — that's marq24's job!. | I live by the motto: Better to ask twice, than to think about it once. | It's marq24's own fault that he provides open-source software and is willing to offer free support.

### Add a description

It's me again ...
After I managed it to access to KNX busses within a single HA instance, I wanted to setup my two SENEC devices accordingly. Knowning that is was working, when I switched my office to Home Assistant.
Unluckily, it's not working anymore... (WebAPI; lala.cgi is working)

As far as my current investigation is:
-> _app_master_plant_id is read from the access-token file in .storage/
-> There's only one file caching the plant information AND DLY/MLY/YLY counters!
     Filename: <Emailaddress>_access_token.txt"
-> Multiple systems accessed using the SAME account (email) cannot work as there's no structural preparation!

I'm currently working on a first "quick" solution:
-> Adding the (selected) plant number to the caching file (<email>_<plantNumber>_access_token.txt)
     The plant number is not added for plant number < 1 (that is -1 "auto", "0" "first plant").

If my approach works, I'll send a pull request to you!
Or do you have another idea?

### Add your DEBUG log output

```shell

```

marq24, 2025-09-15:
hopefully fixed with 2025.9.2

WernerSlabon, 2025-09-16:
2025.9.2 looks good to me.
Now, behavior as intended.

closed 2025-09-16 by marq24